raspiblitz/raspiblitz #606: XXcleanUp.sh - add resetBlockchain option

rootzoll, 2019-05-15:
In some cases people want to trigger a reset on the blockchain data. Add that option to the XXcleanup.sh script and start the "get blockchain" script again and see that it offers reboot when done.

btc21cluib, 2019-05-16:
How can I add the XXcleanup.sh script ?

bitsam, 2019-05-16:
+1 , Thanks @rootzoll  that is what I was asking for in issue  #596

rootzoll, 2019-05-19:
@btc21cluib correction: the script is calles XXcleanHDD.sh and is already part of RaspiBlitz. If you call it, it cleans everything from the HDD (also the LND data), but is keeping the blockchain data ... so you can use the HDD for a fresh RaspiBlitz setup, without the need to download the blockchain data.

What we taking about here is to extend the functionality to also use the script to just delete the blockchain data. That will be part of a future release.

rootzoll, 2019-06-17:
Added a `-blockchain` parameter to `XXcleanHDD.sh` - that will delete all the blockchain data and trigger the new `XXrepairBlockchain.sh`. To be final tested.

rootzoll, 2019-07-01:
Tests look good. This is part of v1.3 release.

closed 2019-07-01 by rootzoll