raspiblitz/raspiblitz #608: Connecting LND node to Mobile Wallet

Fellane, 2019-05-16:
Is it possible to connect my raspiblitz node to any mobile wallet(Zap, Zeus) if my node is runnig through TOR?

CommanderPoe, 2019-05-17:
As far as im concerned if you open the ports you are good to go

openoms, 2019-05-17:
No open ports are needed when on Tor. Just use the menu options.

Fellane, 2019-05-17:
> 
> 
> No open ports are needed when on Tor. Just use the menu options.

I want to connect mobile application to my node from global internet, not from my local network.

openoms, 2019-05-17:
Correction: @CommanderPoe is right, you need to forward the port 8080 (the REST API port) and 22 to be able to ssh in from the outside to be able to use the Zeus wallet remotely: https://twitter.com/openoms/status/1098562853483802624?s=19

Having the node behind Tor does not make difference in accessing ssh, REST or RPC, but you don't need to forward the ports 8333 (bitcoind) and 9735 (lightning) to have your node publicly accessible (for Tor nodes).

Fellane, 2019-05-19:
> 
> 
> Correction: @CommanderPoe is right, you need to forward the port 8080 (the LND REST API port) and 22 to be able to ssh in from the outside to use the Zeus wallet remotely: https://twitter.com/openoms/status/1098562853483802624?s=19
> 
> Having the node behind Tor does not make a difference in accessing ssh, REST or RPC, but you don't need to forward the ports 8333 (bitcoind) and 9735 (lightning) to have your node publicly accessible (for other Tor nodes).
> 
> If you don't have a static IP it is best to set up a dynamic DNS.
> Could also test ZeroTier which sets up a VPN service to connect to your machines from anywhere like you would use the local network:
> #601
Thx for your help.
How can i setup dynamic DNS for raspberry PI? Is there any guide?

rootzoll, 2019-05-19:
@Fellane start in the README for dynamic DNS: https://github.com/rootzoll/raspiblitz#dynamicdns

Fellane, 2019-05-24:
> 
> 
> @Fellane start in the README for dynamic DNS: https://github.com/rootzoll/raspiblitz#dynamicdns

I've successfully installed DDNS and connected Zeus and Zap wallets to my node. All is working fine and i have access to my node through global internet. But there is one problem. I can create invoice but it seems like nobody can pay this invoice. I've made few spins on lightningspin and tried to withdraw my sats but i couldn't. I tried with yalls and also coulnd't get my sats. Can you help me with that problem, please?

openoms, 2019-05-24:
You will need some inbound liquidity to be able to get paid. Please have a look at some of the resources on the end of this readme to understand how it works: https://github.com/openoms/lightning-node-management/blob/master/README.md

closed 2019-05-27 by rootzoll