raspiblitz/raspiblitz #610: Stuck at "The Litecoind service is not running"

coldb, 2019-05-16:
I am trying to set up a Litecoin on RaspiBlitz but getting stuck after creating the LND wallet and rebooting. This is the second time trying but getting stuck with the same issue

when using the **top** command I can see litecoind running. And processing the blockchain I would think.

Is there another litecoind starting before the service and locking the files? Or is this normal behaviour?

**Debug log:**

blitzversion: 1.2
chainnetwork: litecoin / main
 20:35:39 up 9 min,  2 users,  load average: 3.55, 2.78, 1.53

*** BCLOCKHAIN SYSTEMD STATUS ***
● litecoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/litecoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2019-05-16 20:35:39 BST; 418ms ago
  Process: 14749 ExecStart=/usr/local/bin/litecoind -daemon -conf=/home/bitcoin/.litecoin/litecoin.conf (code=exited, status=1/FAILURE)
  Process: 14746 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 14749 (code=exited, status=1/FAILURE)
   CGroup: /system.slice/litecoind.service
           └─5154 /usr/local/bin/litecoind -daemon -conf=/home/bitcoin/.litecoin/litecoin.conf

May 16 20:35:39 ltc systemd[1]: litecoind.service: Failed with result 'exit-code'.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u litecoind -b --no-pager -n8
-- Logs begin at Thu 2019-05-16 20:25:42 BST, end at Thu 2019-05-16 20:35:40 BST. --
May 16 20:35:39 ltc systemd[1]: litecoind.service: Service hold-off time over, scheduling restart.
May 16 20:35:39 ltc systemd[1]: Stopped Bitcoin daemon.
May 16 20:35:39 ltc systemd[1]: Starting Bitcoin daemon...
May 16 20:35:39 ltc systemd[1]: Started Bitcoin daemon.
May 16 20:35:39 ltc litecoind[14749]: Error: Cannot obtain a lock on data directory /mnt/hdd/litecoin. Litecoin Core is probably already running.
May 16 20:35:39 ltc systemd[1]: litecoind.service: Main process exited, code=exited, status=1/FAILURE
May 16 20:35:39 ltc systemd[1]: litecoind.service: Unit entered failed state.
May 16 20:35:39 ltc systemd[1]: litecoind.service: Failed with result 'exit-code'.
1558035310 ERROR: Error found in Logs
1558035328 ERROR: Error found in Logs

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/litecoin/debug.log
2019-05-16 19:35:30 UpdateTip: new best=e87cd7ace7aec6f8fe35de905ea93e3e3a0a1939f9ed67a1f0b17db47add49b9 height=1529945 version=0x20000000 log2_work=72.575638 tx=29271557 date='2018-11-20 06:29:34' progress=0.969515 cache=4.0MiB(35434txo)
2019-05-16 19:35:31 UpdateTip: new best=936fbfacaed006720bafa237124f938ea4d1cb60b53ccf9c66fe9fc221dbada4 height=1529946 version=0x20000000 log2_work=72.575645 tx=29271708 date='2018-11-20 06:37:56' progress=0.969516 cache=4.2MiB(37508txo)
2019-05-16 19:35:31 UpdateTip: new best=2140932ce9a55317c4c3fd8531105d09d7b485f53e432c378ed1821bcd038db7 height=1529947 version=0x20000000 log2_work=72.575652 tx=29271714 date='2018-11-20 06:38:17' progress=0.969516 cache=4.2MiB(37514txo)
2019-05-16 19:35:31 UpdateTip: new best=09624ae2d2be51acef2e5bdbbb6506aef77ebeb6114e3c103dd589c14aee731a height=1529948 version=0x20000000 log2_work=72.575659 tx=29271723 date='2018-11-20 06:38:27' progress=0.969516 cache=4.3MiB(37989txo)
2019-05-16 19:35:31 UpdateTip: new best=325492e53cfcb91892d6f555e4ed9a9dc85542b2e5b2bc9b09321b3053a172c7 height=1529949 version=0x20000000 log2_work=72.575666 tx=29271749 date='2018-11-20 06:40:39' progress=0.969517 cache=4.4MiB(39006txo)
2019-05-16 19:35:32 UpdateTip: new best=611b6031f2445d2afec699b27307cd6c24f3d588205e7d67797227a138f8aac9 height=1529950 version=0x20000000 log2_work=72.575673 tx=29271766 date='2018-11-20 06:40:44' progress=0.969517 cache=4.4MiB(38754txo)
2019-05-16 19:35:32 UpdateTip: new best=7a5161287e1ff0d06b544c2dd900484e5731e3d6d6e7b79b4a4ac08adda96bfe height=1529951 version=0x20000000 log2_work=72.57568 tx=29271769 date='2018-11-20 06:40:56' progress=0.969517 cache=4.4MiB(38658txo)
2019-05-16 19:35:32 UpdateTip: new best=c3cf4147a75c5d4fd6fcf800ec8435962be379971806f66e9a97359315c83c68 height=1529952 version=0x20000000 log2_work=72.575688 tx=29271791 date='2018-11-20 06:42:38' progress=0.969517 cache=4.4MiB(38633txo)
2019-05-16 19:35:32 UpdateTip: new best=bd1d722f35c8cc7c3475325a2a27a106b421af38fb14be649af276eda4aab147 height=1529953 version=0x20000000 log2_work=72.575695 tx=29271828 date='2018-11-20 06:44:50' progress=0.969517 cache=4.4MiB(38506txo)
2019-05-16 19:35:33 UpdateTip: new best=7129df4387eadf1ac1d2c9c7911875021bb35d5946d70887358977b706a001e2 height=1529954 version=0x20000000 log2_work=72.575702 tx=29271875 date='2018-11-20 06:47:26' progress=0.969518 cache=4.3MiB(38459txo)
2019-05-16 19:35:33 UpdateTip: new best=130e6f70f56b07909bbc48f3a2b95655b1ee571c9dd4d5af5b6fb9a11e26e54c height=1529955 version=0x20000000 log2_work=72.575709 tx=29271922 date='2018-11-20 06:50:11' progress=0.969518 cache=4.3MiB(38405txo)
2019-05-16 19:35:33 UpdateTip: new best=200557d45cf54c93ad7f61e14b0e470d34866b37d8284f48572c5a9a8730f4b3 height=1529956 version=0x20000000 log2_work=72.575716 tx=29271984 date='2018-11-20 06:53:05' progress=0.969518 cache=4.3MiB(38346txo)
2019-05-16 19:35:34 UpdateTip: new best=ba6736a8ff40230d338e5850bf799165047611e5573f0deb434a3df14b46cc33 height=1529957 version=0x20000000 log2_work=72.575723 tx=29272030 date='2018-11-20 06:55:27' progress=0.969519 cache=4.3MiB(38260txo)
2019-05-16 19:35:34 UpdateTip: new best=9ea37ec8eb2dc763ac084e120b2b8f4f57f321479b2c203fedbbb7960816f005 height=1529958 version=0x20000000 log2_work=72.57573 tx=29272035 date='2018-11-20 06:56:48' progress=0.969519 cache=4.3MiB(38267txo)
2019-05-16 19:35:34 UpdateTip: new best=52d76e02a0a291e54480726856b6da97240e8f2527e6f7f75b4d835ee87fa594 height=1529959 version=0x20000000 log2_work=72.575737 tx=29272075 date='2018-11-20 06:58:02' progress=0.969519 cache=4.3MiB(38362txo)
2019-05-16 19:35:34 New outbound peer connected: version: 70015, blocks=1633057, peer=10
2019-05-16 19:35:35 UpdateTip: new best=fe8b9dd81bd1600ac5c729a721edeec1cde9723cd4671d4d3936870f76dee4f4 height=1529960 version=0x20000000 log2_work=72.575744 tx=29272130 date='2018-11-20 07:00:47' progress=0.969519 cache=4.3MiB(38398txo)
2019-05-16 19:35:36 UpdateTip: new best=aae5273ed8f385a5b1bba12c7e59b4464842749c942ec1819bbc2f65b92c4d86 height=1529961 version=0x20000000 log2_work=72.575751 tx=29272168 date='2018-11-20 07:04:48' progress=0.969520 cache=4.3MiB(38363txo)
2019-05-16 19:35:36 UpdateTip: new best=2b703b4916cec3e9a0e426d46c3cd40ee6d86b827b48383e5bcfc3934165d99f height=1529962 version=0x20000000 log2_work=72.575758 tx=29272222 date='2018-11-20 07:05:22' progress=0.969520 cache=4.4MiB(38517txo)
2019-05-16 19:35:37 UpdateTip: new best=ceae2fb534a28f0c5315071d52dfe5f007ac0b954302a4d658cce91c108738aa height=1529963 version=0x20000000 log2_work=72.575765 tx=29272242 date='2018-11-20 07:06:55' progress=0.969520 cache=4.3MiB(38207txo)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-05-16 20:29:34 BST; 6min ago
  Process: 5111 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 5116 (lnd)
   CGroup: /system.slice/lnd.service
           └─5116 /usr/local/bin/lnd --externalip=90.190.104.152:9735

[... 51 lines cut ...]
tempWARN=0
tempMAX='6120 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.118'
tempCelsius='59.6'
uptime=590
startcountBlockchain=13
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
* OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=1
lndActive=1
lndRPCReady=1
walletOpened=1
walletReady=1
walletLocked=0
lndErrorShort='Unkown Error - see logs'
lndErrorFull='[lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory'
* OK: log 'ERROR: [lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory' written to /home/admin/systemd.lightning.log
lndRPCReady=1
scriptRuntime=1

coldb, 2019-05-17:
Tried to kill the service and process and restart with the following command 

> sudo systemctl stop litecoind.service && sudo kill -9 $(pidof litecoind) && sudo rm /mnt/hdd/litecoin/.lock && sudo systemctl start litecoind.service

but still the same issue with litecoind it logs

> ltc litecoind[14749]: Error: Cannot obtain a lock on data directory /mnt/hdd/litecoin. Litecoin Core is probably already running.

If I am interpreting the scripts correctly I seem to be stuck in the file 80scanLND.sh at around line 31

Also noticed that lnd seems to be looking for the macaroon in the wrong folder

fluidvoice, 2019-05-18:
as of 19:35:37 in the log it looks like `litecoind` is running and syncing and progress is close to 97% complete. what happens if you just wait longer?

coldb, 2019-05-18:
Nothing happens when it gets to 100%. The debug log looks exactly like above except the blockchain is synced 100%. The screen shows "The litecoind service is not running. Login for more details."

Just let it sync to 100% again and no change: https://termbin.com/wo7l

fluidvoice, 2019-05-18:
what does this show?:
`ps -u bitcoin`

coldb, 2019-05-18:
admin@ltc:~ $ ps -u bitcoin
  PID TTY          TIME CMD
 3025 ?        00:00:05 lnd
 3068 ?        00:46:35 litecoind

fluidvoice, 2019-05-18:
`cat /etc/systemd/system/litecoind.service`

coldb, 2019-05-18:
admin@ltc:~ $ cat /etc/systemd/system/litecoind.service
[Unit]
Description=Bitcoin daemon
Wants=bootstrap.service
After=bootstrap.service

# for use with sendmail alert (coming soon)
#OnFailure=systemd-sendmail@%n

[Service]
User=bitcoin
Group=bitcoin
Type=simple
ExecStartPre=-/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED
ExecStart=/usr/local/bin/litecoind -daemon -conf=/home/bitcoin/.litecoin/litecoin.conf
KillMode=process
Restart=always
TimeoutSec=120
RestartSec=30

[Install]
WantedBy=multi-user.target

fluidvoice, 2019-05-18:
`cat /home/admin/raspiblitz.info`
`cat /mnt/hdd/raspiblitz.conf`

fluidvoice, 2019-05-18:
looks like it getting the config wrong from a file, because I think it should be using "chain/litecoin/mainnet" here instead of "chain/bitcoin/mainnet"
> open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon

coldb, 2019-05-18:
admin@ltc:~ $ cat /home/admin/raspiblitz.info
state=stresstest
message='Testing Hardware 60s'
network=litecoin
chain=main
setupStep=100
undervoltageReports=0

admin@ltc:~ $ cat /mnt/hdd/raspiblitz.conf
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.2'
network=litecoin
chain=main
hostname=ltc
publicIP='90.190.104.152'
autoPilot=off
autoNatDiscovery=off
runBehindTor=off
rtlWebinterface=off
lndPort='9735'
lndAddress=''

fluidvoice, 2019-05-18:
`cat /home/admin/systemd.blockchain.log`

coldb, 2019-05-18:
admin@ltc:~ $ cat /home/admin/systemd.blockchain.log
SYSTEMD RESTART LOG: blockchain (bitcoind/litecoind)
1558207078 STARTED
1558207109 STARTED
1558207139 STARTED
1558207170 STARTED
1558207194 ERROR: Error found in Logs
1558207200 STARTED
1558207215 ERROR: Error found in Logs
1558207230 STARTED
1558207236 ERROR: Error found in Logs
1558207257 ERROR: Error found in Logs
1558207261 STARTED
1558207278 ERROR: Error found in Logs
1558207291 STARTED
1558207299 ERROR: Error found in Logs
1558207321 ERROR: Error found in Logs
1558207322 STARTED
1558207342 ERROR: Error found in Logs
1558207352 STARTED
1558207363 ERROR: Error found in Logs

this continues on for hundreds more lines

fluidvoice, 2019-05-18:
still seems like it must be a config file problem:
`cat /home/admin/.litecoin/litecoin.conf`
`cat /home/admin/.lnd/lnd.conf`

AngelicInsight, 2019-05-18:
@fluidvoice @rootzoll  I'm also experiencing similar symptoms after upgrading to v1.2. The previous version was working perfectly. I've looked through the various config files and they appear to be correct. Something about this update disrupted Litecoin users.

coldb, 2019-05-18:
admin@ltc:~ $ cat /home/bitcoin/.litecoin/litecoin.conf
# litecoind configuration

# mainnet/testnet
testnet=0

# litcoind options
server=1
daemon=1
txindex=1
disablewallet=1

# Connection settings
rpcuser=raspibolt
rpcpassword=****
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

# litecoin specials
discardfee=0.00000001
mintxfee=0.00000001
minrelaytxfee=0.00000001
datadir=/mnt/hdd/litecoin

admin@ltc:~ $ cat /home/bitcoin/.lnd/lnd.conf
# lnd configuration

[Application Options]
debuglevel=debug
maxpendingchannels=5
alias=ltc
color=#68F442
nat=false

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# REST open to all connections on Port 8080
restlisten=0.0.0.0:8080

# Domain, could use https://freedns.afraid.org
#tlsextradomain=lightning.yourhost.com
tlsextraip=0.0.0.0

[Litecoin]
litecoin.active=1
litecoin.mainnet=1
litecoin.node=litecoind

[Litecoind]
litecoind.rpchost=127.0.0.1
litecoind.rpcuser=raspibolt
litecoind.rpcpass=***
litecoind.zmqpubrawblock=tcp://*:28332
litecoind.zmqpubrawtx=tcp://*:28333

[autopilot]
autopilot.active=0
autopilot.maxchannels=5
autopilot.allocation=0.6`

also in the Lightcoin Lighning telegram channel someone just mentioned the following

> I finally got to updating my setup to v1.2
> I’m having the same symptoms as you now. Something was broken in the update for Litecoin setups. 
> It’s definitely looking for files in the wrong paths.

fluidvoice, 2019-05-18:
> @fluidvoice @rootzoll I'm also experiencing similar symptoms after upgrading to v1.2. The previous version was working perfectly. I've looked through the various config files and they appear to be correct. Something about this update disrupted Litecoin users.

Are you using Litecoin?  if not, open a new issue and provide your logs.

AngelicInsight, 2019-05-18:
Yes, using Litecoin. @coldb was referring to me from the Telegram channel.

Logs here: [https://termbin.com/85gj](url)

fluidvoice, 2019-05-23:
> Yes, using Litecoin. @coldb was referring to me from the Telegram channel.
> 
> Logs here: [https://termbin.com/85gj](url)

Let's see this:
`sudo cat /var/logs/syslog | nc termbin.com 9999`

coldb, 2019-05-24:
> > Yes, using Litecoin. @coldb was referring to me from the Telegram channel.
> > Logs here: [https://termbin.com/85gj](url)
> 
> @coldb and @AngelicInsight Let's see this:
> `sudo cat /mnt/hdd/litecoin/debug.log | nc termbin.com 9999`
> `sudo cat /var/log/syslog | nc termbin.com 9999`

@fluidvoice  Had the raspberry turned off so the blockchain is still syncing at the time of writing. But nothing has changed the last times when sync is completes.

`sudo cat /mnt/hdd/litecoin/debug.log | nc termbin.com 9999` ->https://termbin.com/ezqe
`sudo cat /var/log/syslog | nc termbin.com 9999` -> https://termbin.com/xcr3

AngelicInsight, 2019-05-26:
sudo cat /mnt/hdd/litecoin/debug.log | nc termbin.com 9999 ->https://termbin.com/a4cg
sudo cat /var/log/syslog | nc termbin.com 9999 -> https://termbin.com/f27m

fluidvoice, 2019-05-27:
@coldb perhaps this is a clue. It looks like a "Dialog" component is missing from the system, so the install/config is not completely correct.  Dunno why.

In another working (bitcoin) system the log shows this:

May 13 08:51:54 Rektlesscouk _bootstrap.sh[674]: *** Checking Log Size ***
May 13 08:51:54 Rektlesscouk _bootstrap.sh[674]: OK - logs are at 58 MB - within safety limit
May 13 08:51:59 Rektlesscouk _bootstrap.sh[674]: -> sanity check of IP data:
May 13 08:51:59 Rektlesscouk _bootstrap.sh[674]: OK IPv4

but in your log it shows:

> RaspiBlitz systemd[1]: Started User Manager for UID 1000.
> May  3 12:58:37 RaspiBlitz _bootstrap.sh[517]: fbi: no process found
> May  3 12:58:37 RaspiBlitz _bootstrap.sh[517]: Resetting the InfoFile: /home/admin/raspiblitz.info
> **May  3 12:58:37 RaspiBlitz _bootstrap.sh[517]: *** Checking Log Size ***
>May  3 12:58:37 RaspiBlitz _bootstrap.sh[517]: OK - logs are at 2 MB - within safety limit
> May  3 12:58:38 RaspiBlitz _bootstrap.sh[517]: debconf: unable to initialize frontend: Dialog**
> May  3 12:58:38 RaspiBlitz _bootstrap.sh[517]: debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
> May  3 12:58:38 RaspiBlitz _bootstrap.sh[517]: debconf: falling back to frontend: Readline
> May  3 12:58:38 RaspiBlitz _bootstrap.sh[517]: debconf: unable to initialize frontend: Readline
> May  3 12:58:38 RaspiBlitz _bootstrap.sh[517]: debconf: (This frontend requires a controlling tty.)
> May  3 12:58:38 RaspiBlitz _bootstrap.sh[517]: debconf: falling back to frontend: Teletype
> May  3 12:58:41 RaspiBlitz _bootstrap.sh[517]: Creating SSH2 RSA key; this may take some time ...
> May  3 12:58:41 RaspiBlitz _bootstrap.sh[517]: 2048 SHA256:7tXiuQwR1iCcltHoMTKrRLFa/5YcICpsSYrZEfZNEIA root@RaspiBlitz (RSA)
> May  3 12:58:41 RaspiBlitz _bootstrap.sh[517]: Creating SSH2 ECDSA key; this may take some time ...
> May  3 12:58:41 RaspiBlitz _bootstrap.sh[517]: 256 SHA256:ud4CxoVd1/Ziqi6ABFwU2CUINS3vpVqJMLejcaurOBI root@RaspiBlitz (ECDSA)
> May  3 12:58:41 RaspiBlitz _bootstrap.sh[517]: Creating SSH2 ED25519 key; this may take some time ...
> May  3 12:58:41 RaspiBlitz _bootstrap.sh[517]: 256 SHA256:RZG5FCGz7bSE3kIEyCVU9zfipEsRvcnJTFZhUP3NAiw root@RaspiBlitz (ED25519)
> May  3 12:58:42 RaspiBlitz systemd[1]: Reloading.

fluidvoice, 2019-06-07:
It's too difficult to debug/tell what's going on without the device in front of me. 
The error about macaroon path points to possible bad config file or file/directory ownership rights.

> ERROR: [lncli] unable to read macaroon path (check the network setting!): open /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

fluidvoice, 2019-06-07:
> If everything else is working should I be concerned about the litecoind.service errors?

probably not, unless you have the time/interest to investigate it (searching the LND github issues?)

coldb, 2019-06-07:
I can try and make the raspiblitz available over the web for you. Would that be of help?

I wiped the drive and will do the setup again to get to the state described at the start of the issue.

rootzoll, 2019-06-11:
TODO: Make Testrun with Litecoin for next v1.3 release.

rootzoll, 2019-07-01:
Found some spots that made problems when running litecoin - v1.3 should now have working Litecoin support again.

closed 2019-07-01 by rootzoll