raspiblitz/raspiblitz #611: LND disabled, 00infoLCD.sh runs 80scanLND

pg024, 2019-05-16:
Hello, I have disabled the LND service through systemctl.

On the raspberry LCD display it looks like that 80scanLND.sh is running, showing SSH credentials and the remainder "LND is not running" whereas it should run 00infoBlitz.sh instead maybe?

**Workaround**
Comment out this par on 00infoLCD.sh .

`  if LND is syncing or scanning`
 ` lndSynced=$(sudo -u bitcoin /usr/local/bin/lncli --chain=${network} --netw$`
 `  if [ ${lndSynced} -eq 0 ]; then`
`     /home/admin/80scanLND.sh`
  `   sleep 20`
`    continue`
`  fi`

openoms, 2019-05-17:
Hi, sorry it is not clear what are you trying to achieve here.
Re-enable LND and the issue is solved.

fluidvoice, 2019-05-19:
If your goal is to make running LND optional, that is not the goal of this Raspiblitz project, so you'll have to learn all about the scripts and modify them yourself.  Please close this issue.

closed 2019-05-19 by rootzoll