raspiblitz/raspiblitz #612: RTL

livestradamus, 2019-05-17:
I'm trying to install/run RTL from the main menu but it doesn't work.
Does the RTL needs to access tor? The torproject is completely blocked where I am, is there a work around?
Install failure below.

```
Err:5 https://deb.torproject.org/torproject.org stretch/main Sources
  gnutls_handshake() failed: Error in the pull function.
Ign:6 https://deb.torproject.org/torproject.org stretch/main armhf Packages
Ign:7 https://deb.torproject.org/torproject.org stretch/main all Packages
Ign:8 https://deb.torproject.org/torproject.org stretch/main Translation-en
Ign:9 https://deb.torproject.org/torproject.org stretch/main Translation-en_GB
Reading package lists... Done
W: The repository 'https://deb.torproject.org/torproject.org stretch Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch https://deb.torproject.org/torproject.org/dists/stretch/main/source/Sources  gnutls_handshake() failed: Error in the pull function.
E: Some index files failed to download. They have been ignored, or old ones used instead.
Error executing command, exiting
Reading package lists... Done
Building dependency tree
Reading state information... Done
nodejs is already the newest version (8.11.1~dfsg-2~bpo9+1).
0 upgraded, 0 newly installed, 0 to remove and 23 not upgraded.

FAIL - Was not able to install nodeJS 11
ABORT - RTL install
```

openoms, 2019-05-17:
Install manually following these steps: https://github.com/ShahanaFarooqui/RTL/blob/master/README.md#installation

Your current nodejs version should work without the update.

livestradamus, 2019-05-17:
the manual steps worked

closed 2019-05-17 by livestradamus