raspiblitz/raspiblitz #613: Stuck on Node is syncing

DCress12, 2019-05-17:
All was working fine for about one week but now stuck on Node is Syncing for about a week.

Node is Syncing(1)
Blockchain Progress : 99.80%
Lightning Progress 99.89%

Theses percentage have not changed. I have restarted my raspiblitz but that has not helped.

fluidvoice, 2019-05-18:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-do-i-generate-a-debug-report

DCress12, 2019-05-28:
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 16:13:07 up 18:09,  2 users,  load average: 0.72, 0.38, 0.25

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-05-27 22:06:00 BST; 18h ago
  Process: 5204 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 5197 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 5325 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─5325 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitco…id

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- No entries --

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-05-28T15:12:55Z connected to self at 192.168.40.1:63418, disconnecting
2019-05-28T15:12:56Z connected to self at 192.168.40.1:64880, disconnecting
2019-05-28T15:12:57Z connected to self at 192.168.40.1:59979, disconnecting
2019-05-28T15:12:57Z connected to self at 192.168.40.1:62979, disconnecting
2019-05-28T15:12:58Z connected to self at 192.168.40.1:64699, disconnecting
2019-05-28T15:12:58Z connected to self at 192.168.40.1:50343, disconnecting
2019-05-28T15:12:59Z connected to self at 192.168.40.1:63346, disconnecting
2019-05-28T15:13:00Z connected to self at 192.168.40.1:64697, disconnecting
2019-05-28T15:13:00Z connected to self at 192.168.40.1:59413, disconnecting
2019-05-28T15:13:01Z connected to self at 192.168.40.1:54015, disconnecting
2019-05-28T15:13:02Z connected to self at 192.168.40.1:50384, disconnecting
2019-05-28T15:13:02Z connected to self at 192.168.40.1:50370, disconnecting
2019-05-28T15:13:03Z connected to self at 192.168.40.1:58676, disconnecting
2019-05-28T15:13:04Z connected to self at 192.168.40.1:50199, disconnecting
2019-05-28T15:13:04Z connected to self at 192.168.40.1:62580, disconnecting
2019-05-28T15:13:05Z connected to self at 192.168.40.1:64478, disconnecting
2019-05-28T15:13:05Z connected to self at 192.168.40.1:50950, disconnecting
2019-05-28T15:13:06Z connected to self at 192.168.40.1:64484, disconnecting
2019-05-28T15:13:06Z connected to self at 192.168.40.1:50006, disconnecting
2019-05-28T15:13:07Z connected to self at 192.168.40.1:54774, disconnecting

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-05-27 22:10:49 BST; 18h ago
  Process: 22247 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 22250 (lnd)
   CGroup: /system.slice/lnd.service
           └─22250 /usr/local/bin/lnd --externalip=************

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
 MaxChannelSize: (int64) 16777215,
 Private: (bool) false,
 MinConfs: (int32) 0
})

2019-05-27 22:11:04.892 [DBG] LTND: Populating dependencies for sub RPC server: SignRPC
2019-05-27 22:11:04.892 [DBG] LTND: Populating dependencies for sub RPC server: WalletKitRPC
[... 53 lines cut ...]
scanProgress=99.60
lndRPCReady=1
scriptRuntime=0

*** Hardware Test Report ***

RaspiBlitz Hardwaretest v0.1
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
33.8'C 1400 MHz  1.3688V
40.8'C 1400 MHz  1.3688V
42.4'C 1400 MHz  1.3688V
42.9'C 1400 MHz  1.3688V
43.5'C 1400 MHz  1.3688V
44.0'C 1400 MHz  1.3688V
39.7'C 1400 MHz  1.3688V
36.5'C  600/1400 MHz 1.3688V
35.4'C  600/1400 MHz 1.3688V
36.5'C  600/1400 MHz 1.3688V
35.4'C  600/1400 MHz 1.3688V
35.4'C  600/ 600 MHz 1.3688V
34.9'C  600/ 600 MHz 1.3688V
35.9'C  600/1400 MHz 1.3688V
34.9'C  600/ 600 MHz 1.3688V
?1049l
?1049l

rootzoll, 2019-06-03:
Can you check your bitcoin.conf if there is a "addNode" entry with 192.168.40.1 or similar?
See https://bitcointalk.org/index.php?topic=2802418.0

Jos3xTv, 2019-06-10:
Same happened to me and no such entry. This is my bitcoin.conf file.

> # bitcoind configuration

# mainnet/testnet
testnet=0

# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=1

# Connection settings
rpcuser=raspibolt
rpcpassword=passwordB
rpcport=8332
rpcallowip=127.0.0.1
rpcbind=127.0.0.1:8332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

datadir=/mnt/hdd/bitcoin

Jos3xTv, 2019-06-11:
I close Issue. I had to reboot twice to bring back to life my node.

closed 2019-06-11 by rootzoll