raspiblitz/raspiblitz #614: Czech shopping list: adaptor problem

jooray, 2019-05-17:
I bought most of the components from the UK shopping list, instead of the power adapter, because the UK has UK power plug. I wanted to order from German amazon, but they don't ship the power adaptor to Slovakia, so I ordered [the one](https://www.alza.cz/EN/alzapower-q100-quick-charge-3-0-black-d5324257.htm?o=2) from [the Czech shopping list](https://github.com/rootzoll/raspiblitz/blob/master/shoppinglist_cz.md).

When I boot it with a hard drive attached, I get power supply errors during self diagnostic (both warning and critical) consistently. I did not go through.

Do you know of any power adapter that would work and could be ordered either on alza.cz or on Amazon, but that can be shipped outside of Germany/Austria?

Also, I think it might be better to power the whole thing by powered USB hub, so each port would get their own 3A and they would not be fighting for one. Does anyone have any idea which works best?

Anyway, the adapter from the czech shopping list is probably not good enough (or my hard drive needs too much power).

jooray, 2019-05-17:
Maybe @geco91 or @karliatto would know, as they updated the czech shopping list?

amcitaly, 2019-05-18:
In fact the 1.1 version was using the USB hub, but from the 1.2, maybe as the PS is more powerful the guide said to attache the HDD directly on the RaspiBlitz.

fluidvoice, 2019-05-18:
> Does anyone have any idea which works best?

this has been answered many times. Search through the past issues here.

rootzoll, 2019-05-19:
I can change shopping lists based on user recommendations. Often shopping lists from other countries are a first draft and need to be optimized overtime by experience.

HELP WANTED: So it would be a great contribution if someone from Czech can order several Power Supplies and recommend one that is working. Just post here - I will be happy to add to shopping list.

karliatto, 2019-05-20:
@jooray In my original [PR](https://github.com/rootzoll/raspiblitz/commit/fe575b553b2889c6cf590ca4ca767529661fcdd1) the "Power Addaptor" was https://www.alza.cz/EN/raspberry-pi-2-5a-d4191252.htm?o=1, I have that one and it works. Then it was changed to the one you mentioned, I have no experiences with that one.

rootzoll, 2019-05-27:
I switched back to the old power supply link - but made a comment if someone finds a better option

closed 2019-05-27 by rootzoll