raspiblitz/raspiblitz #615: Setup HDD check improvments

CandleHater, 2019-05-18:
```
*** Checking if HDD is connected ***
OK - HDD found at sda1
OK - HDD found at sda2 ... determine which is bigger
sda1(209715200)
sda2(1073741824)
sda2 is BIGGER - run with this one

*** HDD Size Check ***
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
WARNING: HDD might be too small
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
You HDD was detected with the size of 1073741824 bytes
For bitcoin at least 450000000000 bytes is recommended
If you want to change to a bigger HDD:
* Unplug power of RaspiBlitz
* Make a fresh SD card again
* Start again with bigger HDD
If you want to try with HDD connected, press ENTER to continue.
```

- there was a sda3 available with 1.8Tb. It had partitions already, but maybe it should ask if those can be deleted.
- "Make a fresh SD card again" after removing the partitions of sda3 and rebooting, the setup continued and greeted me with "HDD HAS NO PARTITIONS - Press ENTER to create a Partition". So it might (will report) not be needed to make a fresh SD card.
- I think it should convert bytes to Gb/Tb for displaying, so it is clearer to the user.

rootzoll, 2019-05-19:
I added the info that the HDD may need to be re-partitioned first for now.
Somebody who connects a a HDD with 3 partitions should be able to handle this manually.

To start with a fresh SD card in this case is just to play it safe - thats a way it should always work.

TODO for future development: Let the RaspiBlitz make a deeper investigation into the HDD and offer auto re-partitioning.

rootzoll, 2020-04-01:
Doing fresh partitions of HDD/SSD is part of the v1.4 release. Closing issue.

closed 2020-04-01 by rootzoll