raspiblitz/raspiblitz #616: Password limitation to strict

CandleHater, 2019-05-18:
The special char limitation is more strict then necessary. I would like to at least set a more complex user password as I want to use the static password of my Yubikey (on the end; I use it as a "salt") which contains special chars.

rootzoll, 2019-05-19:
This was done because some users reported problems with scripts when using special chars. So to be safe the user is advised to use no special chars for now - longer password can compensate for security.

TODO for later version: make scripts and storage safe to work with all special chars allowed by LND

CandleHater, 2020-01-22:
Until then I'll continue to ``passwd`` to change the user password to a more complex one manually.

rootzoll, 2020-01-29:
Yeah sorry for pushing this issue into the future, but its one of those things when not done right it can break and loose funds quickly .. and at the moment its working.