raspiblitz/raspiblitz #617: Kernel errors

Smiggel, 2019-05-18:
I am copying a downloaded blockchain to my new node (pi3) with new SD flashed card. And during the cloning of the full blockchain from an other powered HDD, I suddenly got kernel errors when the copy process was at 47%.

`Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.391174] Internal error: Oops - undefined instruction: 0 [#1] SMP ARM

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.520446] Process kswapd0 (pid: 44, stack limit = 0xb9f6a210)

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.528152] Stack: (0xb9f6be88 to 0xb9f6c000)

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.533405] be80:                   00000000 000020e8 b9f6bf34 80c85780 00000001 00000000

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.543354] bea0: 00000001 00000000 80c85780 8023bd74 00000000 0000124a 80236b08 00000000

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.553319] bec0: 00000000 00000001 00000000 80d1369c 80d11ee0 80c85780 b9f6bf7c b9f6bee8

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.563304] bee0: 8023bc18 802840e4 00000001 80c20250 80b8cd50 b9e95a00 80c89aa8 80c03d68

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.573289] bf00: 80160e18 b9f6bee8 00000003 b9f6bf34 80c85e3c 80cbeba4 b9f6a000 00000000

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.583347] bf20: 00000000 00000000 00000000 014000c0 00000003 00000000 00000000 0000000c

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.593383] bf40: 00000000 00000007 00000000 00000000 b9d9145c b9d91440 00000000 b9f50c00

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.603440] bf60: 80c85780 8023b948 b9d9145c b9cf9e30 b9f6bfac b9f6bf80 8013de24 8023b954

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.613447] bf80: 80102d94 b9f50c00 8013dce8 00000000 00000000 00000000 00000000 00000000

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.623408] bfa0: 00000000 b9f6bfb0 801080ac 8013dcf4 00000000 00000000 00000000 00000000

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.633381] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.643356] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000

Message from syslogd@RaspiBlitz at May 18 13:04:20 ...
 kernel:[ 8688.681397] Code: e8bd4000 e3510000 e1a08000 e1a0a002 (e50b304c) 
`

fluidvoice, 2019-05-18:
If you have two disk ext4 partitions named BLOCKCHAIN mounted (say on /dev/sdb1 and /dev/sdc1) you can use this to clone from source to destination.

make sure the destination drive is empty first:
`sudo rm -rf /mounted/distination/path/BLOCKCHAIN1/*`

clone the drive with `rsync` (note: you must use a trailing backslash "/" on the source):
`sudo rsync -rv -pogx --update --progress /mounted/source/path/BLOCKCHAIN/ /mounted/destination/path/BLOCKCHAIN1`

Smiggel, 2019-05-18:
Did the sync again and now it worked fine. I removed the lcd on top of the Pi. I think it may have been a heating issue. However, I am not sure.

fluidvoice, 2019-05-18:
> I think it may have been a heating issue

doubtful. More likely your power supply is weak.  Search the past issues here for details on that.

closed 2019-05-19 by rootzoll