raspiblitz/raspiblitz #621: Raspiblitz crashed my external HDDs?

StyriaTrading, 2019-05-20:
I have the following problems with two external HDDs after using them with my Pi/Raspiblitz - one is a Seagate, one an Western Digital:

Everything worked fine for several days with the HDDs, downloaded the whole Blockchain, used Raspiblitz as it was intended to use.

After some days the first HDD could not be mounted again by the Pi and was not recognized anymore. The same thing for the second HDD which I tried after the first.

So I tried the HDDs on my Computer (Linux Mint newest version). The plugin is recognized, but no HDD partitions are displayed, some programs like gparted and R-Linux are recognizing the HDDs but cannot access them --> Read only mode. 

Terminal says:
root@admin:~# dosfsck -a /dev/sda
fsck.fat 4.1 (2017-01-24)
open: Read-only file system

I just want to format it and want to use it again, but nothing is possible. I have no idea how to fix it. Please help me.

fluidvoice, 2019-05-20:
verify the drive is seen as /dev/sda:
`lsblk`
make a primary partition:
`sudo parted -s /dev/sda unit s mkpart primary`
format the partition as ext4 format and name the partition "BLOCKCHAIN":
`sudo mkfs.ext4 /dev/sda -F -L BLOCKCHAIN`

StyriaTrading, 2019-05-20:
The problem is that the HDD is in READ ONLY Mode.

It is not listed under lsblk, but it is definitely running on the PC and also detected in GParted. This has nothing to do with the PSU.

I cannot create any partition because it is in read only mode - and it was also not possible to remove this read only mode.

fluidvoice, 2019-05-20:
> The plugin is recognized

What does this mean?  Do you mean plugging in the HDD it gets recognized, or something else?

> The problem is that the HDD is in READ ONLY Mode.

yes, I understood that.  
Do you have any idea how the drives might have been formatted with NTFS?  Did you attempt to initialize them with a Windows system?... or maybe that error message is not accurate.

do you get any output from running these?:
`sudo gdisk -l /dev/sda`
`sudo parted /dev/sda print`

StyriaTrading, 2019-05-20:
> What does this mean? Do you mean plugging in the HDD it gets recognized, or something else?

I mean that the computer gives the sound output, which occurs, when you plugin an external device - so the harddisk is definitely running (also the LED is blinking).

> yes, I understood that.
> Do you have any idea how the drives might have been formatted with NTFS? Did you attempt to initialize them with a Windows system?... or maybe that error message is not accurate.

Before I used the drives on my Pi with Raspiblitz, they were formatted as NTFS - I think Raspiblitz formats it to EX4, am I right?  I never used them on another device after I started to use them with Raspiblitz. Several days after they simply stopped to working and did not boot again with Raspiblitz.

> do you get any output from running these?:
> `sudo gdisk -l /dev/sda`
> `sudo parted /dev/sda print`

Will try that when I am at home.

StyriaTrading, 2019-05-20:
> do you get any output from running these?:
> `sudo gdisk -l /dev/sda`
> `sudo parted /dev/sda print`

Got following outputs:

xxx@admin:~$ sudo gdisk -l /dev/sda
[sudo] password for julian:           
GPT fdisk (gdisk) version 1.0.3

Partition table scan:
  MBR: MBR only
  BSD: not present
  APM: not present
  GPT: not present

***************************************************************
Found invalid GPT and valid MBR; converting MBR to GPT format
in memory. 
***************************************************************

Warning! Secondary partition table overlaps the last partition by
712261472 blocks!
You will need to delete this partition or resize it in another utility.
Disk /dev/sda: 3200237761 sectors, 1.5 TiB
Model: USB Flash Disk  
Sector size (logical/physical): 512/512 bytes
Disk identifier (GUID): 6C81B5DF-695E-42AC-93A6-8BB8223F6F93
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 3200237727
Partitions will be aligned on 2048-sector boundaries
Total free space is 2014 sectors (1007.0 KiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048      3912499199   1.8 TiB     8300  Linux filesystem

xxx@admin:~$ sudo parted /dev/sda print
Warning: Unable to open /dev/sda read-write (Read-only file system).  /dev/sda
has been opened read-only.
Warning: Unable to open /dev/sda read-write (Read-only file system).  /dev/sda
has been opened read-only.
Error: Can't have a partition outside the disk!
Ignore/Cancel? Ignore                                                     
Model: Generic USB Flash Disk (scsi)
Disk /dev/sda: 1639GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      1049kB  2003GB  2003GB  primary

fluidvoice, 2019-05-21:
did you try this from Windows?:

> Connect the ”Read Only” storage device such as an SD card, USB, external hard drive  to your PC
> Open a Command Prompt by typing: command in the Search tab, click Command Prompt
> Type: **diskpart** and hit Enter
> Type: **list disk** and hit Enter
> Type: **select disk 0** and hit Enter (0 will be the drive letter of your read-only hard drive/USB/SD card, etc.)
> Type: **attributes disk clear readonly** and hit Enter
> Type: **exit** and hit Enter to finish the process

fluidvoice, 2019-05-21:
> did you try this from Windows? ***NOTE*** make sure that you select the correct disk with "select disk #" command below.  Seek out documentation on using `diskpart` if you're unsure so you don't trash your computer's HDD.
> 
> > Connect the ”Read Only” storage device such as an SD card, USB, external hard drive  to your PC
> > Open a Command Prompt by typing: command in the Search tab, click Command Prompt
> > Type: **diskpart** and hit Enter
> > Type: **list disk** and hit Enter
> > Type: **select disk 0** and hit Enter (0 will be the drive letter of your read-only hard drive/USB/SD card, etc.)
> > Type: **attributes disk clear readonly** and hit Enter
> > Type: **exit** and hit Enter to finish the process

StyriaTrading, 2019-05-21:
> did you try this from Windows? _**NOTE**_ make sure that you select the correct disk with "select disk #" command below. Seek out documentation on using `diskpart` if you're unsure so you don't [...]

I tried that already several times, the problem is, that it shows an error in diskpart that the read only attribute cannot be removed.

fluidvoice, 2019-05-22:
> Warning! Secondary partition table overlaps the last partition by
> 712261472 blocks!
> You will need to delete this partition or resize it in another utility

I think this is a big clue.  When I googled this came up on top. Might help: https://unix.stackexchange.com/questions/208256/warning-secondary-partition-overlaps-the-last-partition-by-x-blocks-in-gdisk

StyriaTrading, 2019-05-22:
Thank you for the help - I will try it the next days! Will close the issue for now.

closed 2019-05-22 by StyriaTrading