raspiblitz/raspiblitz #622: Display info page as img file on pi LCD

zanderisrael, 2019-05-20:
Hey Guys, 

Wondering if we can create png files and display them on the LCD with a 5-10 sec refresh rate without using up too much resources.

The Idea of using **chromium** to do the layout is easiest but a bit expensive resource wise.

I think that installing **imagemagick** in order to create the image file (rendering the text in a predefined template image) and then display the image on the lcd using **fbi**, could work.

Does anyone have experience with using **imagemagick** on a pi? is it resource intensive?

raumi75, 2019-05-20:
That's possible to use imagemagick and I've run a few tests. We already use fbi to display QR-codes. Take a look at scripts like these https://github.com/rootzoll/raspiblitz/blob/master/home.admin/XXdisplayLCD.sh 

I was surprised that Imagemagick with fbi was a more resource intensive than simply running chromium or epiphany. I've started writing svg-files that chromium can display in an xserve session. We can take a look at this issue

https://github.com/rootzoll/raspiblitz/issues/88#issuecomment-467578776

in which we discuss how to setup a browser.

Issue #388 has a mock up of a status page based on svg. This is just a proof of concept and I'm not sure if it's worth the effort and the computing power on this little computer.

@frennkie and @fluidvoice were involved, too.

raumi75, 2019-05-20:
This is my latest svg:

[image]

zanderisrael, 2019-05-21:
> That's possible to use imagemagick and I've run a few tests. We already use fbi to display QR-codes. Take a look at scripts like these https://github.com/rootzoll/raspiblitz/blob/master/home.admin/XXdisplayLCD.sh
> 
> I was surprised that Imagemagick with fbi was a more resource intensive than simply running chromium or epiphany. I've started writing svg-files that chromium can display in an xserve session. We can take a look at this issue
> 
> [#88 (comment)](https://github.com/rootzoll/raspiblitz/issues/88#issuecomment-467578776)
> 
> in which we discuss how to setup a browser.
> 
> Issue #388 has a mock up of a status page based on svg. This is just a proof of concept and I'm not sure if it's worth the effort and the computing power on this little computer.
> 
> @frennkie and @fluidvoice were involved, too.

Does anyone have some benchmarks on the pi? using chromium and svg images to render the status screen? I can definitely make it look badass as long as the work is not for nothing.

frennkie, 2019-05-21:
will look into it

rootzoll, 2019-05-21:
related issues for Infoservice and Touch:
#83 #88

rootzoll, 2019-05-28:
Looks like X server is running with less mem foot print as assumed in the beginning - option is back on the table.

frennkie, 2019-06-02:
I still really like the HTML Info Page. But I still would not run a Browser on the RaspiBlitz. Instead the HTML Info Page will be hosted by **blitd** and can be access from
* a desktop/labtop PC
* a mobile phone
* other kinds of devices.

@arcbtc had the idea to use a really small device (ESP32?! Node-MCU? RPi Zero?!..) with a small to medium sized display and use that to show the RaspiBlitz Info Page (physically separated..).

nyxnor, 2021-04-18:
pls close this, Web UI for now and no further improvements recently on this topic.

closed 2021-04-19 by rootzoll