raspiblitz/raspiblitz #625: Several attempts to make payments have failed

lucbran, 2019-05-26:
Hello
todaty tried to pay a Bitrefill invoice but after several attempts gett error (suggesting use RTL).
Problem is also RTL doesnt work... here is log file:
*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
powerFAIL=0
powerWARN=0
powerMIN='1262500 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='6340 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.99'
tempCelsius='62.3'
uptime=2331
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=577882
initialSync=0
syncProgress=100.00 
startcountLightning=3
lndActive=1
walletLocked=0
syncedToChain=1
scanTimestamp=1558882256
scanDate='Sun May 26 15:50:56 BST 2019'
scanProgress=99.99
lndRPCReady=1
scriptRuntime=1

rootzoll, 2019-05-27:
LND failing to execute a payment is not a RaspiBlitz issue - please let the LND people know to get more info on it. If there is anything that needs to be improved on RaspiBlitz side, please report back:
https://dev.lightning.community

lucbran, 2019-05-27:
Hi Christian
tried again today and it worked...may be the problem was Bitrefill Node!! Thanks
Keep up the good job!!

closed 2019-05-27 by lucbran