raspiblitz/raspiblitz #626: The bitcoind service is not running [v1.2]

escapedcat, 2019-05-26:
History of this node. I startet with v1.1. It was running with some `low voltage` warnings.  
I (physically) moved the node to a new location. After turning it on again the `bitcoind` service issues started.  
I updated the power supply and upgraded to v1.2. 

Now I'm stuck with this:  

Status is
```
   ┌── Blockchain Info ──────────
   │  The bitcoind service is not running.          
   │  Login for more details:                                
   │  Use CTRL+c to EXIT to Terminal                
   └────────────────────
```

Somehow is shows this sometimes:
```
*****************************************
* The bitcoind service is not running.
*****************************************
If you just started some config/setup, this might be OK.

59 RESTARTS DETECTED - bitcoind might be in a error loop
1558884610 ERROR: Error found in Logs

More Error Detail:
error code: -28 error message: Loading block index...

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now
```

Output og `XXdebugLogs.sh` is:
```
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 16:30:55 up  1:06,  2 users,  load average: 1.42, 1.33, 1.08

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: signal) since Sun 2019-05-26 16:30:49 BST; 5s ago
  Process: 4534 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 4528 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 4550 (code=killed, signal=ABRT)

May 26 16:30:49 myblitz systemd[1]: bitcoind.service: Unit entered failed state.
May 26 16:30:49 myblitz systemd[1]: bitcoind.service: Failed with result 'signal'.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Fri 2019-05-17 22:17:01 BST, end at Sun 2019-05-26 16:30:55 BST. --
May 26 16:29:42 myblitz systemd[1]: bitcoind.service: Failed with result 'signal'.
May 26 16:30:12 myblitz systemd[1]: bitcoind.service: Service hold-off time over, scheduling restart.
May 26 16:30:12 myblitz systemd[1]: Stopped Bitcoin daemon.
May 26 16:30:12 myblitz systemd[1]: Starting Bitcoin daemon...
May 26 16:30:12 myblitz systemd[1]: Started Bitcoin daemon.
May 26 16:30:49 myblitz systemd[1]: bitcoind.service: Main process exited, code=killed, status=6/ABRT
May 26 16:30:49 myblitz systemd[1]: bitcoind.service: Unit entered failed state.
May 26 16:30:49 myblitz systemd[1]: bitcoind.service: Failed with result 'signal'.
1558884610 ERROR: Error found in Logs
1558884650 ERROR: Error found in Logs

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-05-26T15:30:12Z Default data directory /home/bitcoin/.bitcoin
2019-05-26T15:30:12Z Using data directory /mnt/hdd/bitcoin
2019-05-26T15:30:12Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-05-26T15:30:12Z Using at most 40 automatic connections (1024 file descriptors available)
2019-05-26T15:30:12Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-05-26T15:30:12Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-05-26T15:30:12Z Using 4 threads for script verification
2019-05-26T15:30:12Z scheduler thread start
2019-05-26T15:30:12Z HTTP: creating work queue of depth 16
2019-05-26T15:30:12Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2019-05-26T15:30:12Z HTTP: starting 4 worker threads
2019-05-26T15:30:12Z Cache configuration:
2019-05-26T15:30:12Z * Using 2.0MiB for block index database
2019-05-26T15:30:12Z * Using 12.2MiB for transaction index database
2019-05-26T15:30:12Z * Using 8.0MiB for chain state database
2019-05-26T15:30:12Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-05-26T15:30:12Z init message: Loading block index...
2019-05-26T15:30:12Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2019-05-26T15:30:12Z Opened LevelDB successfully
2019-05-26T15:30:12Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-05-26 15:26:47 BST; 1h 4min ago
  Process: 2179 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 2183 (lnd)
   CGroup: /system.slice/lnd.service
           └─2183 /usr/local/bin/lnd --externalip=46.59.252.111:9735

May 26 15:26:47 myblitz systemd[1]: Starting LND Lightning Daemon...
May 26 15:26:47 myblitz systemd[1]: Started LND Lightning Daemon.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Fri 2019-05-17 22:17:01 BST, end at Sun 2019-05-26 16:30:55 BST. --
May 26 15:26:47 myblitz systemd[1]: Starting LND Lightning Daemon...
May 26 15:26:47 myblitz systemd[1]: Started LND Lightning Daemon.

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2019-05-26 15:26:49.746 [INF] LTND: Version: 0.6.0-beta commit=v0.6-beta, build=production, logging=default
2019-05-26 15:26:49.746 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-05-26 15:26:49.882 [INF] CHDB: Checking for schema update: latest_version=8, db_version=8
2019-05-26 15:26:49.928 [INF] RPCS: password RPC server listening on [::]:10009
2019-05-26 15:26:49.939 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-05-26 15:26:49.940 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

*** LAST 20 RTL LOGS ***
-- Logs begin at Fri 2019-05-17 22:17:01 BST, end at Sun 2019-05-26 16:30:55 BST. --
May 26 15:26:47 myblitz systemd[1]: Started RTL daemon.

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
powerFAIL=0
powerWARN=0
powerMIN='1312500 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='3470 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.178.58'
tempCelsius='48.6'
uptime=4017
startcountBlockchain=59
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=1
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
scriptRuntime=1
```

`sudo tail -f /mnt/hdd/bitcoin/debug.log`  
```
2019-05-26T15:44:26Z HTTP: starting 4 worker threads
2019-05-26T15:44:26Z Cache configuration:
2019-05-26T15:44:26Z * Using 2.0MiB for block index database
2019-05-26T15:44:26Z * Using 12.2MiB for transaction index database
2019-05-26T15:44:26Z * Using 8.0MiB for chain state database
2019-05-26T15:44:26Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-05-26T15:44:26Z init message: Loading block index...
2019-05-26T15:44:26Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2019-05-26T15:44:26Z Opened LevelDB successfully
2019-05-26T15:44:26Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
```

Hardwaretest:
```
     ┌──── Hardware Check ────────
     │                                                                          
     │ RaspiBlitz hardware setup looks good :)      
     │ Your are ready to continue - have fun.         
     │                                                                          
     ├────────────────────
```

Not sure what to do. I searched the current/closed issues but couldn't find something matching. Let me know if I missed something or if I should add more infos to this.  
I'm close to just start everything from 0 again. Might be easier.

fluidvoice, 2019-05-26:
Let's see the whole logs:
`sudo cat /mnt/hdd/bitcoin/debug.log | nc termbin.com 9999`
`sudo cat /var/log/syslog | nc termbin.com 9999`

and your config file:
`sudo cat /home/bitcoin/.bitcoin/bitcoin.conf`

escapedcat, 2019-05-26:
- https://termbin.com/n22d
- https://termbin.com/yw2b

```
# bitcoind configuration

# mainnet/testnet
testnet=0

# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=1

# Connection settings
rpcuser=raspibolt
rpcpassword=
rpcport=8332
rpcallowip=127.0.0.1
rpcbind=127.0.0.1:8332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

datadir=/mnt/hdd/bitcoin
```

fluidvoice, 2019-05-27:
Your installation is broken/mal-formed for some reason. It's missing programs that should be there. For example in the log ...
> _bootstrap.sh[495]: debconf: unable to initialize frontend: Dialog

I would do a fresh install.

escapedcat, 2019-05-27:
Alright, thanks for your time and feedback!  
I'll start fresh and use the recover option for my wallet. Didn't have any active channels.

closed 2019-05-27 by escapedcat

MaxexaM, 2019-09-22:
I have the same problem, re-installed many times, replaced SD card ... everything works, the blockchain starts to download but then suddenly it stops at 8% or 10% ... could it be the SSD?

rootzoll, 2019-10-02:
@MaxexaM check the SSD casing/USB2SATA adapter. Some of a freezing problem not supporting UASP correctly. The one in the shopping lists should work.

MaxexaM, 2019-10-03:
hi Christian, ok, I tried another HDD, 1 week to download the 100% blockchain and then again "bitcoind not running"
https://termbin.com/kvl7

rootzoll, 2019-11-20:
I am not able to access the posted log files (maybe outdated due to mylate reponse). But its not the HDD/SSD the problem I described - its the USB adater hardware that comes with the case that needs to support UASP. Can you post a link to the SSD & Case you are using?