raspiblitz/raspiblitz #627: CPU stats Celsius <-> Fahrenheit conversion bug

vit001, 2019-05-26:
Stats screen is showing CPU temperature 51C / 83F. One of the two values is wrong. Perhaps a unit conversion bug?

rootzoll, 2019-05-27:
This is how it gets calculated ... do you have a more exact way?
```
FAHRENHEIT=$(((CELSIUS) * (9/5) + 32))
```

vit001, 2019-05-27:
The formula is correct. I suspect 9/5 are treated as ints and result in 1,
so the formula reduces to F = C + 32. You may have to do some casting to
float to get the 9/5 handled correctly. I realize this is a very minor
issue :)

Dimpelmoser, 2019-06-02:
... I hope I do not embarrass myself, but I believe this could be a solution for a calculation with integers:
FAHRENHEIT=$((((CELSIUS) * 18) + 320)/10) 
... Please, in which file can I find the current calculation. I would like to try it.

rootzoll, 2019-06-03:
@Dimpelmoser edit file: `00infoBlitz.sh` in your `/home/admin`directory - around line 58 :D

Dimpelmoser, 2019-06-03:
@rootzoll - Respect for your great work

... Since I was inexperienced in LINUX syntax, I have made several(!) attempts.
The following solution has led to a satisfactory result.

tempF = $ ((((cpu / 1000) * (18) + 320) / 10))

But with the little flaw that the calculated and displayed Fahrenheit-result will not be rounded.
Example:
56°C is actually 132.8°F - but 132°F is displayed and not 133°F.
That's just the price for this simplified solution.

For the given purpose, however, it might be enough.

[... I believe that the syntax in LINUX may be optimized by someone who knows better about it than me.]

rootzoll, 2019-06-05:
@Dimpelmoser thanks :) I added to new formula to the code for v1.3
marking this ticket for final testing

Dimpelmoser, 2019-06-08:
... sorry, but I'm an esthete and perfectionist ... sometimes ...

What bothered me with my previous solution was that the result was not decently rounded. That's why I've been looking for a way to solve that - and I think I found one.

With the following formula, the Fahrenheit temperature is displayed correctly rounded on the screen of my little raspiblitz - so from x.5 to the next higher integer ° F.

tempF = $ (((((cpu / 1000) * 18 + 320) + 5) / 10))

rootzoll, 2019-06-11:
This looks good :) I added it to the v1.3 release branch. Thanks.

Dimpelmoser, 2019-06-12:
... and finally the shortest possible alternatives (space-optimized formulas):

1) tempF=$((((cpu / 1000) * 18 + 325) / 10))
or
2) tempF=$(((tempC * 18 + 325) / 10))

Both alternatives show on the LCD-status-screen of my RaspiBlitz the temperature rounded up from x.5° to the next higher integer degree of °Fahrenheit.

rootzoll, 2019-06-12:
Love your optimization effort :) Commited to the v1.3 code. Thanks.
Tests looking good - I will close this issue.

closed 2019-06-12 by rootzoll