raspiblitz/raspiblitz #635: Cant connect with ZAP when on mobile network

adolfosrs, 2019-05-30:
I'm being able to connect remotely to my node using Zap when I'm on a regular wi-fi connection (not local). But I'm not being able to connect when trying it on a mobile 4G network. I'm on a complete different network when trying the wifi approach. 

And I'm also being able to ssh through the 4G connection but when trying to connect using zap the app refuses to connect. 

When trying to scan the lndconnect QR code with my phone already on the mobile network I get the following: 

> Could not connect to server. (Name resolution failure)

I'm currently using a DynamicDNS from http://freedns.afraid.org

rootzoll, 2019-06-03:
Hard to say whats wrong. Maybe head over to the ZAP iOS github and ask how you can get more info on the exact reason? https://github.com/LN-Zap/zap-iOS

HonmaSokyu, 2019-06-06:
Hey!

Ill just ask here. Can someone please point me to the direction "how to connect ZAP with my Node, through Mobile network? 

Is it still not possible, because i have TOR on my Node?

Thank you!

rootzoll, 2019-06-11:
I am closing this issue - but if there is any more info to add for future users, please add still as a comment.

closed 2019-06-11 by rootzoll

adolfosrs, 2019-06-11:
FYI I have opened a new issue on LN-Zap repo. No answer so far.

https://github.com/LN-Zap/zap-iOS/issues/178