raspiblitz/raspiblitz #637: Raspiblitz seems to be freezing

glensidecross, 2019-06-01:
Hi. I'm running the most recent version of Raspiblitz on a Raspberry Pi 3B+.  I've been running it since about October of last year and it's been very smooth.

I connect to it using Zap IOS, and about two weeks ago noticed that the connection was unstable. I would have to try to connect several times before it would go through.  Then, the Rabpiblitz would seem to freeze - the screen would be the same lightning screen, but I couldn't connect with RTL, Zap, or SSH.  When I reboot, it operates normally for a little while, and then same thing. 

At the time this started, Tokyo was having a hot stretch, and when I ran a hardware test it said there were heat warnings. I upgraded to a armor case so the pi is now running cooler.
I just now ran a hardware test and it says everything is normal. I am pasting the results of a software test below.

I love the raspiblitz and appreciate all the hard work being put into it. I hope I will be able to resolve the problems.

Questions:

* Could the 'hot' periods have caused faults in the pi = I should be a new pi?
* Could the card be corrupted? If so should I reburn the card? Will that mess with any data I have?

Thanks for your help!!

fluidvoice, 2019-06-01:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-do-i-generate-a-debug-report

glensidecross, 2019-06-05:
Hi, Thanks for the help.

I have determined that the issue is not heat. The heat warnings have gone away. I seem to be getting undervolt warnings during hardware checks. However, my power supply is a NorthPada 5v 3a unit. 

I actually now think the problems for me started when I upgraded to the latest version of Raspiblitz, which is 1.2.

Here is the debug report. Any help that can be offered would be greatly appreciated.

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 00:12:05 up 19 min,  3 users,  load average: 3.22, 3.26, 2.23

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-06-05 23:55:37 BST; 16min ago
  Process: 2120 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 2113 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 2154 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─2154 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Jun 05 23:55:37 BitcoinWarrior.net systemd[1]: bitcoind.service: Failed to read PID from file /home/bitcoin/.bitcoin/bitcoind.pid: Invalid argument
Jun 05 23:55:37 BitcoinWarrior.net systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Thu 2019-06-06 00:12:05 BST. --
Jun 05 23:55:36 BitcoinWarrior.net systemd[1]: Starting Bitcoin daemon...
Jun 05 23:55:37 BitcoinWarrior.net systemd[1]: bitcoind.service: Failed to read PID from file /home/bitcoin/.bitcoin/bitcoind.pid: Invalid argument
Jun 05 23:55:37 BitcoinWarrior.net systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-06-05T23:00:27Z socket send error Broken pipe (32)
2019-06-05T23:00:40Z socket send error Broken pipe (32)
2019-06-05T23:01:27Z New outbound peer connected: version: 70015, blocks=579427, peer=42
2019-06-05T23:01:28Z New outbound peer connected: version: 70015, blocks=579427, peer=45
2019-06-05T23:01:53Z Imported mempool transactions from disk: 2883 succeeded, 0 failed, 0 expired, 5 already there
2019-06-05T23:01:59Z socket send error Broken pipe (32)
2019-06-05T23:02:07Z New outbound peer connected: version: 70015, blocks=579427, peer=56
2019-06-05T23:02:18Z New outbound peer connected: version: 70015, blocks=579427, peer=60
2019-06-05T23:03:48Z socket send error Broken pipe (32)
2019-06-05T23:04:13Z socket send error Broken pipe (32)
2019-06-05T23:04:48Z socket recv error Connection reset by peer (104)
2019-06-05T23:05:55Z New outbound peer connected: version: 70015, blocks=579427, peer=87
2019-06-05T23:07:04Z socket recv error Connection reset by peer (104)
2019-06-05T23:07:05Z socket send error Broken pipe (32)
2019-06-05T23:07:11Z socket recv error Connection reset by peer (104)
2019-06-05T23:07:29Z socket send error Broken pipe (32)
2019-06-05T23:07:35Z New outbound peer connected: version: 70015, blocks=579428, peer=102
2019-06-05T23:07:43Z socket send error Broken pipe (32)
2019-06-05T23:09:15Z New outbound peer connected: version: 70015, blocks=579430, peer=118
2019-06-05T23:10:53Z socket send error Broken pipe (32)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-06-05 23:55:37 BST; 16min ago
  Process: 2161 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 2164 (lnd)
   CGroup: /system.slice/lnd.service
           └─2164 /usr/local/bin/lnd --tor.active --tor.v2 --listen=127.0.0.1:9735

Jun 05 23:55:37 BitcoinWarrior.net systemd[1]: Starting LND Lightning Daemon...
Jun 05 23:55:37 BitcoinWarrior.net systemd[1]: Started LND Lightning Daemon.

[... 52 lines cut ...]
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.0.16'
tempCelsius='54.7'
uptime=1146
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=579427
initialSync=0
syncProgress=100.00
startcountLightning=1
lndActive=1
walletLocked=0
syncedToChain=1
scanTimestamp=1559774999
scanDate='Wed  5 Jun 23:49:59 BST 2019'
scanProgress=99.99
lndRPCReady=1
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

glensidecross, 2019-06-06:
Hoy. Thought I'd add one more detail. It does seem like LND is freezing. The lightning display still shows, but I can't SSH in until I reboot. I can control+c to get to the terminal and reboot that way.

closed 2019-06-06 by glensidecross

fluidvoice, 2019-06-07:
It doesn't matter what the PS is rated 3A or not, or even if other reviews says it works well. You could have gotten a bad one.  Check your connections and if the low voltage reports continue replace the PS. Otherwise you could get corrupted data on your HDD and other problems.

glensidecross, 2019-06-07:
Understood. Thanks for your help.

I am still having issues with what seems to be the raspiblitz software
freezing. The Pi is still running, though.

Any help you can offer would be appreciated, but even if you are focused on
more important things, know that I am thrilled with Raspiblitz and love the
work you folks are doing.

Mark

fluidvoice, 2019-06-07:
> what seems to be the raspiblitz software freezing. The Pi is still running, though.

I don't see clear evidence anything is wrong, other than your inability to SSH into the node.
It looks like Bitcoin has synced 100% and LND scan also.  
So by "freezing" perhaps this is just a SSH and/or networking problem. 
What happens or what do you see when you attempt SSH login?
Can you ping the IP address of the node from another computer?
When you `CTRL+c` to get to a prompt are you doing that from a keyboard connected to the Pi or from where?

glensidecross, 2019-06-09:
Hi. Thanks again for all your help.

I've spent a few days trying different things.

It seems that the power supply was the issue. My power supply seems strong enough, but I had it plugged into a an extension cord plugged into a power strip. When I rigged it so the pi power supply was plugged directly into the powerstrip, the power warnings dropped from every time I run a hardware test to about 1 in 10.  The Raspiblitz has not frozen since I plugged it directly into the powerstrip.

As a side note, through all of this, my Zap IOS app has been really funky about connecting with the pi. This has also stabilized since I removed the extension cord.

I don't know, but suspect, that the upgrade to Raspiblitz 1.2 increased the power demand so the power configuration I had before was no longer sufficient.

I'm going to keep trying to find ways to make the power supply more stable or stronger.  I really appreciate your taking the time to look at this and respond.

Thanks for all the hard work you're putting in. I can't wait for future versions.