raspiblitz/raspiblitz #638: test LND v0.7.0

fluidvoice, 2019-06-01:
LND v0.7.0 contains a bug fix to avoid the "edge not found" error considered to be a `lnd` database corruption caused by a race-condiition and hit by at least eight Raspiblitz nodes.  It's important we test this release and nodes upgrade to it.  Note: does not fix DB corruption so nodes with this will need to reinstall+restore.

> There is a potential fix in upcoming LND v0.7.0 but it only avoids the problem for working nodes, it does not cure the problem for nodes already with the database corruption:
> https://github.com/lightningnetwork/lnd/issues/3025#issuecomment-497458753

> > We've found a potential deadlock when polling GetInfo that could cause the daemon to not shut down cleanly, which would explain why users are experiencing this database issue.
> > 
> > The deadlock would cause lnd not to shut down completely, so if the process gets force killed then it's possible to run into database corruption. The fix is now in master and will be included in the upcoming v0.7.0 release, so it should prevent nodes with this fix from running into this issue. **For any other nodes without this fix that continue to run into this issue, there's not much we can do other than the recommended recovery process.**
> > 
> > With that said, I'll go ahead and close this for now. Once v0.7.0 is out and most Raspiblitz nodes have upgraded, we can re-examine any further reports if the issue seems to persist.

_Originally posted by @fluidvoice in https://github.com/rootzoll/raspiblitz/issues/562#issuecomment-497939315_

openoms, 2019-06-03:
You can build from source up to the commit of your choice with this script:
https://github.com/openoms/bitcoin-tutorials/blob/master/lnd.from.source.sh

Run:
`$ wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.from.source.sh && sudo bash lnd.from.source.sh`
and it will ask for the commit ID from this list: https://github.com/lightningnetwork/lnd/commits/master.

conejoninja, 2019-06-29:
I run into this issue recently (2 days ago). Not really sure what should I do: wait for new version? try to fix the database corruption? clean the hdd and restart again?

I'm not sure what to delete from the HDD, I would like to avoid downloading the blockchain again if possible. I do have the channels.backup, but is it corrupt? or what is exactly corrupt?

rootzoll, 2019-06-29:
You can try if a update to LND v0.7-RC2 fixes this ... get to Teminal and run: 
```
wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.update.v0.7.0-beta-rc2.sh && sudo bash lnd.update.v0.7.0-beta-rc2.sh
```
After that reboot: `sudo shutdown -r now`

conejoninja, 2019-06-29:
no luck, same message
```
 [ERR] SRVR: unable to start server: edge not found
```

but it worked for a while, so I'm suspecting a hardware issue maybe? A new power supply is coming, but have some doubts about the recovery process. do I need to backup the macaroons and wallet.db/macaroons.db too?

conejoninja, 2019-07-01:
Use the channel.backup to try to recover it but it doesn't, 0 satoshis, 0 channels, no history, everything is empty

openoms, 2019-07-01:
> no luck, same message
> 
> ```
>  [ERR] SRVR: unable to start server: edge not found
> ```
> 
> but it worked for a while, so I'm suspecting a hardware issue maybe? A new power supply is coming, but have some doubts about the recovery process. do I need to backup the macaroons and wallet.db/macaroons.db too?

Sorry to say, but this seems to be an unrecoverable error. It was fixed in a way that it won`t happen after lnd v.0.7.0, but unfortunately there is no  method to repair corrupted no node.
You will need to restore form seed and SCB as described here: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#2-recover-from-wallet-seed and further issues discussed here: https://github.com/rootzoll/raspiblitz/issues/632

openoms, 2019-07-01:
## update to v0.7.0-beta-rc3
https://github.com/openoms/bitcoin-tutorials/tree/master/lnd.updates

Download and run this script on the RaspiBlitz:
`$ wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.updates/lnd.update.v0.7.0-beta-rc3.sh && sudo bash lnd.update.v0.7.0-beta-rc3.sh`

https://twitter.com/openoms/status/1145714613335613440

conejoninja, 2019-07-01:
I deleted the lnd folder and recover with seed + channel.backup. It showing me my wallet balance and my channels are "Waiting Close Channel"
[image]

not sure how to proceed

openoms, 2019-07-01:
> I deleted the lnd folder and recover with seed + channel.backup. It showing me my wallet balance and my channels are "Waiting Close Channel"
> [image]
> 
> not sure how to proceed

Restoring from the Static Channel Backup has successfully triggered a fore close on your channels. Will need to wait a day for the funds to become available as part of the on-chain balance,

openoms, 2019-07-02:
## update to v0.7.0-beta
Run the one-liner to update:
`$ wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/lnd.updates/lnd.update.v0.7.0-beta-rc3.sh && sudo bash lnd.update.v0.7.0-beta-rc3.sh`

https://github.com/openoms/bitcoin-tutorials/tree/master/lnd.updates#update-lnd-to-v070-beta
https://twitter.com/openoms/status/1146159431085899778

rootzoll, 2019-07-02:
OK I was always hoping that 0.7.0 will "fix" LNDs having that error - nit just preventing it for the future. RaspiBlitz v1.3 will have LND 0.7.0 ... so i will close this issue.

closed 2019-07-02 by rootzoll

conejoninja, 2019-07-03:
Updated to v0.7.0 with the on-line without any issues, running fine now. Sorry this might be a bit offtopic, could find much help about the recovery process, Looks like some channels were closed but there's not settled balance and some funds are not appearing on the wallet. Will a re-scan help? looks like it will only with on-chain funds, but this seems to be from channels.

rootzoll, 2019-07-03:
It may take some time until funds get available. You may want to join the telegram channel for assistance: https://t.me/raspiblitz