raspiblitz/raspiblitz #640: Set custom fee in GUI

cryptomulde, 2019-06-02:
Would be great to have a way to set custom fees for funding and closing channels in the GUI, like one can do, by entering `lncli openchannel nodeid 500000 --sat_per_byte 4`

CommanderPoe, 2019-06-03:
I highly support this, even better if it is "compulsory" for users to set their own fees to open/close channels. Standard fees are stupidly high with an empty mempool most of the time. We have to start optimizing these kinds of features by checking the mempool before we do any on-chain tx. Most of the users I speak with whenever we want to open/close a channel don't need this in 10-20 min, waiting 1-2h for a channel or even longer (a day) is pretty plausible unless you actually have an emergency then we can rbf (replace by fees) somehow?. Exchanges are abusing the mempool and we should not contribute to this. IMHO make users to set their own on-chain fees is win-win from every angle.

rootzoll, 2019-06-03:
With GUI do you mean the SSH dialogs or the RideTheLightning webgui?

I would like to keep the SSH dialogs as simple as possible for now - just to point out starting points.

Also whats you opinion on:
https://github.com/bitromortac/lndmanage
it think it could be a place to add all the "channel management" for people that dont want to use the webGUI and use the terminal

cryptomulde, 2019-06-03:
Yes I am referring to the Terminal GUI (ssh dialogs). 

Understood the simplicity argument. RTL already allows setting sat/bytes today.

I first stumbled upon that lndmanage tool yesterday. Looks promising, but I did not check it out yet. Will try running it this week, but I think it can be a useful add to Raspiblitz, also the rebalancing functions look very promising. Rebalancing channels today can not be done easily without a script, hence it would be very nice to include this on board as an optional service.

rootzoll, 2019-06-28:
Due to time issues will put this back to future release planning.

Marcoblue, 2020-09-25:
This Issue is still open. Can it be closed or is changing the milestone an option?

Anyhow I read here about RBF (Replace by Fee). How can I use it? I have a pending payment for a channel opening. It is pending 31h now.

joernroeder, 2020-09-27:
just came across this and i am wondering why this could not be an option in e.g. `lnd.conf`. so users can change it and it might be less work.

openoms, 2020-09-28:
RBF cannot be used to facilitate the confirmation of channel open and close transactions the same way as for a single-sig bitcoin transaction.

There is a `BumpFee` RPC call, but not available for `lncli`: https://api.lightning.community/#bumpfee

For closing transactions in the menu LND's automatic fee estimation is used and I think that is suitable since this is a fallback option and aimed to have minimal complexity.

nyxnor, 2021-04-18:
Can this be closed?

rootzoll, 2021-04-19:
With v1.7 the `lnd.conf` can now be edited from the SSH menu `SYSTEM)`. So I think its fair to close this issue for now.

closed 2021-04-19 by rootzoll