raspiblitz/raspiblitz #642: LND Locked after update from v1.1 to v1.2

inpharmaticist, 2019-06-03:
Debug output:
https://termbin.com/22p9

fluidvoice, 2019-06-04:
Did you unlock the wallet? Did you wait for the blockchain sync to finish?

inpharmaticist, 2019-06-04:
Yes, it says it unlocked, then asks for password again, then says it may be stuck in a loop.

[image]

It may be because I tried to install EPS and the guide had me make some edits to the bitcoin and lnd config files. EPS didn't work (newb). But the node still worked until I tried to update raspiblitz.

fluidvoice, 2019-06-04:
wait until bitcoin is fully synced, then unlock the wallet.  If it still doesn't work run this:
`sudo cat /mnt/hdd/bitcoin/debug.log | nc termbin.com 9999`
`sudo cat /var/log/syslog | nc termbin.com 9999`

inpharmaticist, 2019-06-04:
Bitcoin is synced, I did run that and pasted it in my first comment. Here it is again:

Debug output:
https://termbin.com/22p9

inpharmaticist, 2019-06-05:
I now notice the bitcoin.conf file is empty. Not sure if the v1.2 update did this. Can the node operate if this file is blank?

rootzoll, 2019-06-05:
The bitcoin.conf being empty is a bug - to be fixed in next version. Please fil lthe file again with data from this template with adding your personal rpcpassword (password B):
https://github.com/rootzoll/raspiblitz/blob/master/home.admin/assets/bitcoin.conf

fluidvoice, 2019-06-07:
> Bitcoin is synced, I did run that and pasted it in my first comment. Here it is again:
> 
> Debug output:
> https://termbin.com/22p9

I didn't ask you to re-send me the same log output again. I asked you to send me new logging.
Additionally I asked you to send me the contents of your `syslog`.
But anyway, it looks like the problem is the empty bitcoin.conf file.  But that might be just a symptom of something else wrong.

inpharmaticist, 2019-06-07:
@fluidvoice My bad, didn't realize you needed an updated log. @rootzoll that seemed to do the trick. Thanks both of you!

closed 2019-06-07 by inpharmaticist