raspiblitz/raspiblitz #644: DynamicDNS vs. 1ML

wuestenigel, 2019-06-04:
Hey guys,

after the great BlockChainHotel I'm building more and more stuff with my RaspiBlitz. I even connected it to my Zap iOS Wallet! thx @rootzoll!

Question: I also setup a dynamic DNS domain:

[image]

Nevertheless, on 1ML.com I only see the IP adress. Is there a way to "tell" 1ML.com they should show the subdomain?

Link: https://1ml.com/node/036120a3e610da6a20350559bc2b23cfe2d844f02ff3c8fe41c8718da8d83d0541

CommanderPoe, 2019-06-05:
No. You node is running on clearnet hence publicly reachable, and that'sall the information 1ml prints ( yout public IP) Your DDNS is only useful for wallets (like zap) to alert whenever that IP changes it gets auromatically updated by that dynamic domain you set. However, you can always claim your node on 1ml dox yourself and put all the info you want ( just like you did)

wuestenigel, 2019-06-05:
Thanks for the clarification @CommanderPoe.

So the responsibility for a correct connection is on side of the other node owner?

If he makes the "mistake" and just copy/paste the connection code, it may happen that after a few days my IP changes and then the peer is dead?

Instead I should encourage people in the description to use my DDNS instead? Will this update automatically if my IP changes or is the IP only analyzed once and then stored as IP value on the other node?

rootzoll, 2019-06-05:
If you make a `lncli getinfo` on the terminal, you can see that even if you have a DDNS or another fixed domain set, that LND is not announcing an IP as the URI. 

RaspiBlitz is starting LND with the `--externalip=` parameter - but even when there a domain is given like "raspiblitz.com:9377" LND will resolve to IP and use that to announce URI.

@wuestenigel I was not able yet how to make LND to announce the domain yet. If you can research that or ask on the LND Developer Slack, that would be great.

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll