raspiblitz/raspiblitz #646: Test Touch Contribution

rootzoll, 2019-06-05:
test pull request https://github.com/rootzoll/raspiblitz/pull/645

rootzoll, 2019-07-01:
@frennkie I tried to run thru the installation. On "B4 desktop Autologin" i got a error. For the raspbian buster base image i was choosing this time the "lite" version see: https://www.raspberrypi.org/downloads/raspbian/ .. dou you think the "desktop" verison is needed?

rootzoll, 2019-07-02:
@frennkie I tested with "Raspbian Desktop" and had more success :) 

I got to the point to see after restart the buttons on the left 👍 will see how far I get the rest running.

rootzoll, 2019-07-03:
Got Touchscreen displaying working so far :) Added it service menu so it can be turned on/off. 
In v1.3 it will still be off by default - but to try it out and improve it can be easily switched on.

TODO: Make buttons working
TODO: Check if QR codes getting showed, wehn touchscreen mode is on.

rootzoll, 2019-07-03:
@frennkie found that my buttons were not working, because my touchscreen is very badly calibrated ... I finally found the spot in the sreen that will trigger the shutdown ;) Do you know a way how to calibrate the LCD?

rootzoll, 2019-07-03:
OK QR codes are still working :)
But I still havnt found a way to calibrate the touchscreen.

frennkie, 2019-07-04:
Yes. The `/home/admin/LCD-show/xinput-calibrator_0.7.5-1_armhf.deb` package contains `/usr/bin/xinput_calibrator` and I recall having used it to calibrate my screen some while ago.

So basically install and run:

```
sudo dpkg -i /home/admin/LCD-show/xinput-calibrator_0.7.5-1_armhf.deb
sudo -i -u pi DISPLAY=:0.0 xinput_calibrator
```

This outputs something like this: 

```
Calibrating EVDEV driver for "ADS7846 Touchscreen" id=6
        current calibration values (from XInput): min_x=3936, max_x=227 and min_y=268, max_y=3880

Doing dynamic recalibration:
        Setting calibration data: 3948, 265, 204, 3921
        --> Making the calibration permanent <--
  copy the snippet below into '/etc/X11/xorg.conf.d/99-calibration.conf' (/usr/share/X11/xorg.conf.d/ in some distro's)
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "3948 265 204 3921"
        Option  "SwapAxes"      "1"
EndSection
```

Then create/update `/etc/X11/xorg.conf.d/99-calibration.conf` and restart X (init 3 ; sleep 1 ; init 5) or simply reboot. That should be it.

rootzoll, 2019-07-25:
Even after calibration I was not able to hut the buttons in a reliable way ... hmmm.
So I was always in doubt if the calibratin worked oit ... so I searched for other ways to calibrate and found the tslib: http://www.circuitbasics.com/raspberry-pi-touchscreen-calibration-screen-rotation/
There is also a calibration programm and a even a test programm ... so you can confirm that it was working. 

@frennkie do you think the tslib could be a good repleacement to the one in the LCDshow?

rootzoll, 2019-09-03:
@frennkie the disable of screensaver works great. I also added touchscreen rotation now so that is fits with the rotation set on your RaspiBlitz and buttons work like a breeze now. I think this is no a very cool very to release as experimental for v1.3 this week :D And if that works out great maybe even with next release as standard.

closed 2019-09-03 by rootzoll