raspiblitz/raspiblitz #648: [Question] - Will Adafruit PiTFT 3.5" display work?

denisinla, 2019-06-05:
Currently have an Adafruit PiTFT 3.5" display. Exactly the one below for visuals.

 [image]

After getting everything setup I keep getting greeted with two issues. The text does not render all the way on the screen resulting in a [cut off blue screen](https://d.pr/free/i/MBvJYb).

The second issue being a red message "Screen is not refreshing itself ... please ENTER to continue". I'm assuming this may be because wallet is locked?

---

**Debug Log**
https://termbin.com/e7ci

---

rootzoll, 2019-06-11:
"Screen is not refreshing itself ... please ENTER to continue" is displayed if you run the infoi screen from the main menu. The main menu is not to be displayed on the LCD - its normally just what you have when you login as user admin thru SSH. The LCD should run as user pi - that auto logs in on start and the LCD is just displaying the status information.

So someting maybe wrong with the user running on that screen.

frennkie, 2019-10-03:
@denisinla Could you retry this with the current release v1.3?

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll