raspiblitz/raspiblitz #649: Suggestion: Add a priceticker to the main screen

StyriaTrading, 2019-06-06:
Hello,

just thought about how Raspiblitz would look like with an additionally added price-ticker screen.

For example you can run it like a screensaver, switching between the standard screen and the priceticker. 

Would look really nice in my opinion / would be an eyecatcher.

Thought about looking something similar to this: https://steemit.com/cryptocurrency/@cryptotaylor/raspberry-pi-cryptocurrency-ticker

What do you think? Would it be much effort to include it? Individualization of the settings maybe over the main menu?

rootzoll, 2019-06-11:
That woudl be a nice idea, but at the moment we try to reduce dependencies on central information services as much as possible for default features .. and for price information you always need to contact a central service. So this could be a add-on feature that can be switched-on in the future as a contribution - but its not high on my personal prio list. The staus screen might also be update in the future with more menu options (touch, buttons, epaper, ..) at that point price ticker could be someting to add by user choice.

nyxnor, 2021-04-18:
pls close this. Already reponded on other issue

closed 2021-04-19 by rootzoll