raspiblitz/raspiblitz #650: "Free HDD" output seems incorrect (over 100%)

grnqrtr, 2019-06-08:
The output of "Free HDD" is returning over 70000%

`Free HDD 1018G (72714%)`

Does anyone know what would be causing this error and how to fix it?

grnqrtr, 2019-06-08:
Sorry, I should have tried tinkering around before reporting the issue. I ended up figuring out the problem.

I edited the math on a line in 00infoBlitz.sh. It was trying to divide 1018GB by 1.4TB and was getting messed up.

This was the original line:
`hdd_free_ratio=$(printf "%d" "$(df -h | grep '/mnt/hdd$' | awk '{ print $4/$2*100 }')" 2>/dev/null)`

This is the line edited. I changed `$4/$2*100` to `$4/($2*1000)*100`:
`hdd_free_ratio=$(printf "%d" "$(df -h | grep '/mnt/hdd$' | awk '{ print $4/($2*1000)*100 }')" 2>/dev/null)`

Problem solved!

closed 2019-06-08 by grnqrtr

rootzoll, 2019-06-11:
To make it work with HDDs over 1 TB I changed the script. Please check if once v1.3 gets released that is is still working correctly -thanks.

grnqrtr, 2019-06-13:
Great, thanks! I will check!