raspiblitz/raspiblitz #652: Can't close lightning channels

tryolly, 2019-06-11:
Hi, I'm trying to close my open channels but keep getting an error message (see attached image). My hdd is failing so I want to close all channels before switching to new drive. Any help in closing the channels would be appreciated.
[image]

rootzoll, 2019-06-11:
try the following command:
```
sudo -u bitcoin lncli closeallchannels --force
```

tryolly, 2019-06-11:
will do, but got to go to work now. I'll try in about 9 hours and let you know how it goes.

tryolly, 2019-06-12:
So I got home and there were several errors with the PI, this is no doubt due to hdd. After a very slow shutdown I restarted it and before going to terminal I tried closing the channels from the menu screen again. It worked this time without issue, so yay, time to sort out a new hdd now.

Thanks for the help.

closed 2019-06-12 by tryolly

rootzoll, 2019-06-12:
Make sure to check the power supply (big & stable) is important - better even add a powered usb hub between Raspberry Pi and HDD ... because most HDD problems come from weak power supplies. Good Luck.