raspiblitz/raspiblitz #654: Possible man in the middle attack

kradordep, 2019-06-13:
The display was sayning that 

The Bitcoind service in not runnig

sos when I tried loggin in to my Raspiblitz today I got this:

apples-MacBook-Pro:~ apple$ ssh admin@192.168.0.200
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ECDSA key sent by the remote host is
SHA256:cJgOIKguMNW3Rh6cGcFW20XGswnFF0i0VZur0SNd45Q.
Please contact your system administrator.
Add correct host key in /Users/apple/.ssh/known_hosts to get rid of this message.
Offending ECDSA key in /Users/apple/.ssh/known_hosts:1
ECDSA host key for 192.168.0.200 has changed and you have requested strict checking.
Host key verification failed.
apples-MacBook-Pro:~ apple$

rootzoll, 2019-06-13:
See FAQ foe the beginnung: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#what-to-do-when-on-ssh-i-see-warning-remote-host-identification-has-changed

If you can login again - check why bitcoind is not running and send debug report:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-do-i-generate-a-debug-report

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll