raspiblitz/raspiblitz #655: sudden undervolt issues

glensidecross, 2019-06-16:
I've had a Raspiblitz running since last September fairly flawlessly. It's now running 1.2.

About a month ago, it started acting acting funny - Zap wouldn't connect and it would seeminly freeze.

All of a sudden, I'm getting lots of undervolt warnings.
* My HDD is run from a powered hub.
* I have two 5v 3a powersupplies that I've been switching out, but both of them are showing the issue.
* I live in Japan and the 5v 3a power supplies seem to be the best to be had.
* I wonder if there isn't an error in the pi that's causing the sudden increase in undervolt warnings.

Also, I note that in the image on this github, the CPU usage is .67 or such. Mine is currently 1.84, 1.92, 2.03. Perhaps this is leading to the problem? 

Thanks in advance for any help you can offer!!

rootzoll, 2019-06-17:
> Also, I note that in the image on this github, the CPU usage is .67 or such. Mine is currently 1.84, 1.92, 2.03. Perhaps this is leading to the problem?

That should be OK .. but it can be that because of your power issues your CPU speed gets trottled by RaspberryPi inner logic and so you end up with higher CPU usage in general.

That your HDD is powered by a hub is good to protect against data corruption and may be a hint that there is in general something with your raspberryPi off. Have you tried the change the RaspberryPi board? You can simply power down, exchange board and power up again - should start where it left off.

glensidecross, 2019-06-17:
Hoy! 

Thanks for the comment. I really appreciate it. I have thought that there might be something off with the pi itself, but have been trying all other fixes first.

They aren't that expensive, so...

Love your work and can't wait for future iterations! Keep it up.

closed 2019-06-17 by glensidecross