raspiblitz/raspiblitz #656: Setup Issues After Torrent Bitcoind

anonymoustrident, 2019-06-16:
Hi all , had no issues till after completing the torrent this appears. What i did was to reboot it as per it says but nothing changes.

```
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 17:43:57 up 25 min,  2 users,  load average: 0.56, 0.54, 0.46

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2019-06-16 17:43:51 BST; 6s ago
  Process: 11464 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 11460 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 11470 (code=exited, status=1/FAILURE)

Jun 16 17:43:51 RaspiBlitz systemd[1]: bitcoind.service: Unit entered failed state.
Jun 16 17:43:51 RaspiBlitz systemd[1]: bitcoind.service: Failed with result 'exit-code'.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2019-06-16 17:18:43 BST, end at Sun 2019-06-16 17:43:57 BST. --
Jun 16 17:42:59 RaspiBlitz systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Jun 16 17:43:29 RaspiBlitz systemd[1]: bitcoind.service: Service hold-off time over, scheduling restart.
Jun 16 17:43:29 RaspiBlitz systemd[1]: Stopped Bitcoin daemon.
Jun 16 17:43:29 RaspiBlitz systemd[1]: Starting Bitcoin daemon...
Jun 16 17:43:29 RaspiBlitz systemd[1]: Started Bitcoin daemon.
Jun 16 17:43:51 RaspiBlitz systemd[1]: bitcoind.service: Main process exited, code=exited, status=1/FAILURE
Jun 16 17:43:51 RaspiBlitz systemd[1]: bitcoind.service: Unit entered failed state.
Jun 16 17:43:51 RaspiBlitz systemd[1]: bitcoind.service: Failed with result 'exit-code'.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-06-16T16:43:29Z * Using 2.0MiB for block index database
2019-06-16T16:43:29Z * Using 12.2MiB for transaction index database
2019-06-16T16:43:29Z * Using 8.0MiB for chain state database
2019-06-16T16:43:29Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-06-16T16:43:29Z init message: Loading block index...
2019-06-16T16:43:29Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2019-06-16T16:43:29Z Opened LevelDB successfully
2019-06-16T16:43:29Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2019-06-16T16:43:49Z LoadBlockIndexDB: last block file = 1591
2019-06-16T16:43:49Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=61, size=75273487, heights=570847...570907, time=2019-04-09...2019-04-09)
2019-06-16T16:43:49Z Checking all blk files are present...
2019-06-16T16:43:50Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2019-06-16T16:43:50Z Opened LevelDB successfully
2019-06-16T16:43:50Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: 0000000000000000
2019-06-16T16:43:50Z : Error initializing block database.
Please restart with -reindex or -reindex-chainstate to recover.
2019-06-16T16:43:50Z Aborted block database rebuild. Exiting.
2019-06-16T16:43:50Z Shutdown: In progress...
2019-06-16T16:43:50Z scheduler thread interrupt
2019-06-16T16:43:50Z Shutdown: done

*** LND SYSTEMD STATUS ***
Unit lnd.service could not be found.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

- RTL is OFF by config

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
powerFAIL=0
powerWARN=0
powerMIN='1256300 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='6070 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.XXX.X.XX'
tempCelsius='45.0'
uptime=1510
startcountBlockchain=27
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999
```

anonymoustrident, 2019-06-17:
> Hi all , had no issues till after completing the torrent this appears. What i did was to reboot it as per it says but nothing changes.
>
> ``` [...]

Any Kind Folks can advise on the above issues to proceed on?

rootzoll, 2019-06-17:
Your problem is:
```
2019-06-16T16:43:50Z : Error initializing block database.
Please restart with -reindex or -reindex-chainstate to recover.
2019-06-16T16:43:50Z Aborted block database rebuild. Exiting.
```
So something went bad with your blockchain download. If you havent funded anything I would recommend starting fresh. Format HDD on laptop to FAT32 and make fresh sd card image.

closed 2019-06-17 by rootzoll

anonymoustrident, 2019-06-18:
Thank You, doing it again.