raspiblitz/raspiblitz #657: Missing Bitcoin after closing channels

rpjcbs, 2019-06-16:
Evening all, so I wanted to close my channels and withdraw my BTC from my node.

Turned off autopilot fine and then closed all channels.

Then went to the withdraw Bitcoin option and it said it will withdraw all confirmed Bitcoin, all $22 worth... (5ebe580ff223402cb3b049f2453b0375b31e3edfb85b674dd23a6de960b1f0c0)

I had over $300 sat on there and now there is showing as having nothing... my node address is here https://1ml.com/node/02c0bcae3cc61891e4a0b7635d0c7f14dc1f72a4a0d80ddaa89c689a1c01052f68

If anyone can get their head around what's gone on here I'd be very appreciative.

glensidecross, 2019-06-17:
Not sure about this, but last time I closed channels, it took time for them to actually close and transfer funds to the wallet. I waited up to a week, I believe, but think I got most if not all the sats back.

rpjcbs, 2019-06-17:
@glensidecross that's promising, thanks for your reply, not bothered if I have to wait ages just don't want them to be lost.

rootzoll, 2019-06-17:
If you had a force closing on a channel it will take multiple days until the funds get back to you. Thats the idea of lightning, that there is a waiting period on a "force close" for the other party to check that you really published the latest state on closing. A "cooperative close" is normally a matter of some blocks. 

Another point can be low fees on a closing transaction. Due to price hype (over 9000) there seems a lot transaction in the mempool now waiting: https://jochen-hoenicke.de/queue/#0,4d

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll