raspiblitz/raspiblitz #658: how to generate new wallet?

fuligjimmy, 2019-06-18:
hi,

I forgot to make a note on the words and would love to generate a new wallet, how could i do it?

thx,
fuligjimmy

rootzoll, 2019-06-18:
I would consider a fresh setup. But you can keep the blockchain:
- cash out all your funds and channels
- exit to terminal - run command: `./XXcleanHDD.sh` (it will keep the blockchain data)
- shutdown - run command: `sudo shutdown -r now`
- make a fresh sd card
- and then follow the fresh setup

closed 2019-06-18 by rootzoll

HenrySabio, 2020-05-10:
@rootzoll I didn't want to reopen a new issue for the same thing, but this was a year ago.
Is there a way to create a new wallet without cleaning the drive and a fresh raspiblitz SD build?

rootzoll, 2020-05-11:
To reset the LND wallet check in the SSH Main Menu `REPAIR` > `RESET-LND` ... if you dont see this option, you may need to update the RaspiBlitz release. v1.5 final is now available.

billgre, 2021-05-15:
Hi i received a ready 2 go raspi. I made my Lnd wallet but not an on chain one. Never asked for it.

rootzoll, 2021-05-18:
@billgre The LND is managing both - your on-chain and off-chain funds. By default Bitcoin is just an information service on the blockchain for LND.