raspiblitz/raspiblitz #659: Activate TOR as relay node

rootzoll, 2019-06-18:
In a future version it would be good if people can turn on TOR relay node in the service section. See details on config in this issue comment: https://github.com/rootzoll/raspiblitz/issues/592#issuecomment-502937781

rootzoll, 2020-10-05:
Having this option would fit perfect into the "Make Tor default" goal for v1.7

petrhejna, 2020-12-09:
I was just thinking, that Raspi is becoming decently popular (at least in my bubble). And I have the
 - [ ] run Tor relay node

... on my checklist for some time. :) Really looking forward to it.

rootzoll, 2020-12-10:
I like the idea of activating the node as a Tor relay server in the settings as an option. Just trying to get the details right ... maybe @openoms can join.

So if I take a look at the running relays you can see that they are advertising their IP to the network:
https://metrics.torproject.org/rs.html#toprelays

My takeaway here is that we should not set the same name as the lightning node here for the Tor relay - otherwise we may give up what we tried to hide in the first place. But it should be OK to set a `Nickname` like "raspiblitz736363" (where the number is random) or would that offer an attack vetor?

Also to run a relay does it only makes sense if the router is portforwarding the `ORPort 9001` and `DirPort 9030`set in `/etc/tor/torrc` or would it still work even if the user is not portforwarding from the router to the raspiblitz?

And we should be safe to not running as an Exit Node if we have `ExitPolicy *:*` as part of our config, correct?

openoms, 2020-12-10:
Since the public IP is involved running a relay would need port forwarding same as with https://www.torbox.ch/ https://github.com/radio24/TorBox

I am not comfortable with exposing to the public internet, getting more connections and traffic on a node were there are hot wallets. Same reason why it shouldn't be advertised that it is a RaspiBlitz even if it is activated optionally.

For me I wouldn't use this and would recommend running a dedicated machine for the purpose like the Torbox.

We are doing relaying on Tor with lnd and bitcoind already, it might be better to keep focus.

petrhejna, 2020-12-11:
I was thinking about it more and I must admit that running relay node should be done on different machine. 

I am thinking about similar solution like raspiblitz but with all the good sruff like

- tor
- i2p
- [nym](https://nymtech.net/)
- ...are there others?

rootzoll, 2020-12-14:
maybe pi-hole has an option to run it as a relay tor server?

nyxnor, 2021-04-18:
[Being Tor exit relay is not recommended to be at home](https://community.torproject.org/relay/technical-considerations/). 
[It will also consume more resources from the Pi, from 0,5GB to 1,5GB of RAM.](https://community.torproject.org/relay/relays-requirements/)
[This is the pi-hole option.](https://docs.pi-hole.net/guides/misc/tor/using-tor/)

closed 2021-04-19 by rootzoll