raspiblitz/raspiblitz #663: installing the raspiblitz

vegas555, 2019-06-22:
At the stage of installing the raspiblitz operating system image from a flash drive, installation is not going on. Hangs on this line. Set with 2 different memory cards.
[image]
What could be the problem?

rootzoll, 2019-06-23:
This reminds me of this issue that we still havent found a solution to:
https://github.com/rootzoll/raspiblitz/issues/195

If you have the time it would be great if you can try exchange also the RaspiberryPi board and also try to start up without the HDD. Even trying to start it on another Internet connection. And let us know if that changed anything.

vegas555, 2019-06-25:
Give thanks. Disable HDD. The process has progressed further. But it was still.
[image]

souspeed, 2019-06-28:
I have the same issue.
After working for several months, my raspiblitz now hangs on boot.
Tried version 1.0, 1.1 and 1.2, with and without hdd and network cable. 
Since the raspiblitz doesn’t boot I have no idea what to check further.

souspeed, 2019-06-28:
When the Raspiblitz stopped working I was still on version 1.1. And I'm not able to boot on this version.
With version 1.2 I was able to SSH back into Raspiblitz. (PS I do not have an LCD).
Now it appears that bitcoind and LND do not start automatically, since I cannot get past the hardware test in SSH.

Below is the output for the debug logs. (./XXdebugLogs.sh)

What should I do next?
I have off-chain funds and channel funds on the HDD, created under version 1.1.

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 18:39:16 up 17 min,  2 users,  load average: 0.03, 0.15, 0.22

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- No entries --

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

*** LAST 20 RTL LOGS ***
-- No entries --

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 5
powerFAIL=0
powerWARN=2
powerMIN='1243800 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='6010 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.178.20'
tempCelsius='53.1'
uptime=1060
startcountBlockchain=0
bitcoinActive=0
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
scriptRuntime=0

souspeed, 2019-06-28:
This is the error I get after the hardware test during boot in SSH.

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 20:16:41 up  1:05,  2 users,  load average: 0.95, 0.54, 0.26

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- No entries --

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

*** LAST 20 RTL LOGS ***
-- No entries --

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 5
powerFAIL=0
powerWARN=7
powerMIN='1243800 microVolt'
tempFAIL=0
tempWARN=2
tempMAX='6550 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.178.20'
tempCelsius='56.3'
uptime=3958
startcountBlockchain=0
bitcoinActive=0
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
scriptRuntime=6

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

UNKOWN STATE - there is blockain data config, but blockchain service is not running
It seems that something went wrong during sync/download/copy of the blockchain.
Or something with the config is not correct.
Sometimes a reboot helps --> sudo shutdown -r now

souspeed, 2019-06-29:
Had to manually start bitcoins and LND. This I have to do every reboot of Raspiblitz.
Node is running again.

Still some commands are not working yet, and had to reinstall RTL.
Managed to withdraw all on-chain funds, however still working on the channel funds.

rootzoll, 2019-07-28:
The hanging on boot is a replica issue of #195 - so I am closing this issue. But feels free to report your further progress here.

closed 2019-07-28 by rootzoll