raspiblitz/raspiblitz #664: Systemd complaining about lnd and bitcoind service being executable

frennkie, 2019-06-22:
Not sure whether I made any modifications..  but I don't think so.
```
systemd[1]: Configuration file /etc/systemd/system/lnd.service is marked executable. Please remove executable permission bits. Proceeding anyway.
systemd[1]: Configuration file /etc/systemd/system/bitcoind.service is marked executable. Please remove executable permission bits. Proceeding anyway.
```

In any case this is an easy fix:

```
sudo chmod a-x /etc/systemd/system/bitcoind.service
sudo chmod a-x /etc/systemd/system/lnd.service
```

rootzoll, 2019-06-23:
Hmm never saw that- but will add this to the v1.3 and test it afterwards. Thx.

rootzoll, 2019-06-28:
OK changed. Testing: 
TODO -> Check everything is running after fresh setup
TODO -> Check everything is running after recover

rootzoll, 2019-07-01:
Test look good. This will be part of v1.3 release.

closed 2019-07-01 by rootzoll