raspiblitz/raspiblitz #665: LND service not running | Enters failed state after panic & invalid argument

RanaMX, 2019-06-23:
Hello,

My lightning runs into the following errors which results in a failed state/ error loop:
panic: invalid freelist page: 0, page type is unknown<00> 
lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT 
Complete Errorlog: https://termbin.com/h9t1

Before I fried my Raspiblitz HDD in a reboot due to undervoltage (used 1A outlet from Router :S ) but I could still get the .lnd-rescue-xxx.tar file off the HDD later on.
Now I have a new SSD, reloaded the complete blockchain, but with raspiblitz v1.2 and also with v1.1 and v1.0 my lightning always gets stuck in an error loop.
With v1.0 and v1.1 it hangs in the filtering blockchain "?/5...." stage, with v1.2 it tells me straight away "The LND service is not running."

I have tried to find a solution with other issue# but could not find it. On current v1.2 i have run the following reports, hoping you know what is wrong.
I think worst case my .tar file is corrupted, but since Lightning does not even start I hope it could be another issue.

I currently run on 5V 2.5A for the raspiblitz and apart 5V 1A with Y-cable for the SSD. Hardwaretest says the power supply seems ok, but could be better for stable operations. I will buy another one in the near future.

Complete Errorlog: **https://termbin.com/h9t1**

Hope you can help me!!!
Much thanks!

**Initial Start up v1.2:**

> ****************************************
> * The LND service is not running.
> ****************************************
> If you just started some config/setup, this might be OK.
> 
> 8 RESTARTS DETECTED - LND might be in a error loop
> Jun 23 01:57:38 RanaMX lnd[22811]: panic: invalid freelist page: 0, page type is unknown<00>
> 
> ****************************************
> * The LND service is not running.
> ****************************************
> If you just started some config/setup, this might be OK.
> 
> 8 RESTARTS DETECTED - LND might be in a error loop
> Jun 23 01:57:38 RanaMX lnd[22811]: panic: invalid freelist page: 0, page type is unknown<00>
> Jun 23 01:57:38 RanaMX lnd[22811]: goroutine 1 [running]:
> Jun 23 01:57:38 RanaMX lnd[22811]: github.com/coreos/bbolt.(*freelist).read(0x3044340, 0x63cf6000)
> Jun 23 01:57:38 RanaMX lnd[22811]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/freelist.go:266 +0x23c
> Jun 23 01:57:38 RanaMX lnd[22811]: github.com/coreos/bbolt.(*DB).loadFreelist.func1()
> Jun 23 01:57:38 RanaMX lnd[22811]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/db.go:310 +0xf0
> Jun 23 01:57:38 RanaMX lnd[22811]: sync.(*Once).Do(0x2db40c0, 0x3063d28)
> Jun 23 01:57:38 RanaMX lnd[22811]:         /usr/local/go/src/sync/once.go:44 +0xb8
> 
> -> Use following command to debug: /home/admin/XXdebugLogs.sh
> -> To force Main Menu run: /home/admin/00mainMenu.sh
> -> To try restart: sudo shutdown -r now

**admin@RanaMX:~ $ sudo journalctl -f -u lnd**

> -- Logs begin at Sun 2019-06-23 02:03:43 BST. --
> Jun 23 02:17:19 RanaMX systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
> Jun 23 02:17:19 RanaMX systemd[1]: lnd.service: Unit entered failed state.
> Jun 23 02:17:19 RanaMX systemd[1]: lnd.service: Failed with result 'exit-code'.
> Jun 23 02:18:19 RanaMX systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
> Jun 23 02:18:20 RanaMX systemd[1]: Stopped LND Lightning Daemon.
> Jun 23 02:18:20 RanaMX systemd[1]: Starting LND Lightning Daemon...
> Jun 23 02:18:20 RanaMX systemd[1]: Started LND Lightning Daemon.
> Jun 23 02:18:20 RanaMX systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
> Jun 23 02:18:20 RanaMX systemd[1]: lnd.service: Unit entered failed state.
> Jun 23 02:18:20 RanaMX systemd[1]: lnd.service: Failed with result 'exit-code'.

**admin@RanaMX:~ $ sudo systemctl status lnd**

> ● lnd.service - LND Lightning Daemon
>    Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
>    Active: activating (auto-restart) (Result: exit-code) since Sun 2019-06-23 02:18:20 BST; 46s ago
>   Process: 2607 ExecStart=/usr/local/bin/lnd --externalip=${publicIP}:${lndPort} (code=exited, status=2)
>   Process: 2603 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0
>  Main PID: 2607 (code=exited, status=2)
>    CGroup: /system.slice/lnd.service
> 
> Jun 23 02:18:20 RanaMX systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
> Jun 23 02:18:20 RanaMX systemd[1]: lnd.service: Unit entered failed state.
> Jun 23 02:18:20 RanaMX systemd[1]: lnd.service: Failed with result 'exit-code'.
> lines 1-11/11 (END)

rootzoll, 2019-06-23:
Not sure exactly why ... can you check your raspiblitz config and post the values of `lndPort` and `publicIP` here? --> terminal: `cat /mnat/hdd/raspiblitz.conf`.

Also if your RaspiBlitz was funded/channels and you were running v1.2 ... please check if you can find and backup the following file: `/home/admin/.lnd/data/chain/bitcoin/mainnet/channel.backup`

RanaMX, 2019-06-24:
Hi rootzoll,
unfortunately the raspiblitz was still on v1.1 when HDD got damaged....

Now with the new SSD on v1.2 the cat command does not seem to work, the output is:

> admin@RanaMX:~ $ cat /mnat/hdd/raspiblitz.conf
> cat: /mnat/hdd/raspiblitz.conf: No such file or directory

rootzoll, 2019-06-24:
Sorry there was a typo - it should be: `cat /mnt/hdd/raspiblitz.conf`

RanaMX, 2019-06-25:
Perfect, here comes the output
IP and port seem correct. lndAddress empty.
Starting it up today I also got the message on the raspiblitz screen "Give root password for maintenance or press Ctr+D to continue"
Connecting a USB keyboard and pressing Ctr+D starts it up normal. It was shutdown always via terminal & sudo shutdown now

> admin@RanaMX:~ $ cat /mnt/hdd/raspiblitz.conf
> # RASPIBLITZ CONFIG FILE
> raspiBlitzVersion='1.2'
> network=bitcoin
> chain=main
> hostname=RanaMX
**> publicIP='189.166.43.184'**
> autoPilot=off
> autoNatDiscovery=off
> runBehindTor=off
> rtlWebinterface=off
**> lndPort='9735'**
> lndAddress=''

RanaMX, 2019-07-07:
Well, meanwhile I have done the following:
Changed to 3A power supply and better cooling, very rarely now that I have untervoltage warnings in the logs.
Made a complete new installation from scratch as a test and it started up just fine, so Hardware appears to be fine
Got a new recovery gz.tar file from a new 1.2 SD installation from failing HDD
Installed 1.2 again from scratch on SSD with new .gz.tar file and same result =(
https://termbin.com/rr9u

I still have the assumption that maybe the gz.tar file is corrupted since the old HDD is failing, despite the fact that I can download it without failures from old HDD.

Will try another new recovery installation when v1.3 comes out and if it does not work recover with seedphrase and forget about the onchain funds....

RanaMX, 2019-07-07:
Just tried to access the lightning channels with the Lightning Desktop App, but also unsuccessful. When I exchange the files in the lnd folder of AppData the Lightning Desktop App starts up and stays in the Main screen just saying LIGHTNING By Lightning Labs, INC, not advancing any further.
Also one additional doubt, by now I have pulled 3 different rescue files and all have different names and especially file sizes, is this normal?
[image]

fkochdev, 2019-07-17:
I have the same problem. LND service not running, it enters a failed state after panic.

Tried to clean HDD, copy the blockchain from anonther raspiblitz and tried to restore my old LND node with the .gz.tar backup file. Problem remains.

fluidvoice, 2019-07-22:
I'm running Raspiblitz x86 in a VM and got the same LND failure error.   
> lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT

This was after running perfectly then enabled Tor which was not working, then turning Tor off again and resetting LND via the REPAIR RESET-LND menu selection.  Will followup in another issue:

nyxnor, 2021-05-17:
@RanaMX Please close this issue.

RanaMX, 2021-05-18:
Before closing just the hint for anybody reading this in the future: I could never recover the channels with the raspiblitz, I ended up using guggero's chantools to recover sats from some of the LN channels.

closed 2021-05-18 by RanaMX

nyxnor, 2021-05-18:
@RanaMX Chantools was later added to v1.7 https://github.com/rootzoll/raspiblitz/pull/1962