raspiblitz/raspiblitz #666: Watchtower service

openoms, 2019-06-23:
Got the altruistic watchtower function working on my RaspiBlitz:
https://github.com/openoms/lightning-node-management/blob/master/watchtower.md 
https://twitter.com/openoms/status/1142739780947664896

It is great to monitor for breaches if someone running more than one node.
Needs to store all previous channels states (couple of hundred bytes each) of the monitored clients so will keep checking the diskspace usage.

Could not make it work as a Tor hidden service yet: https://twitter.com/openoms/status/1142759272079667201

I would be a great menu addition for a future version to set up a watchtower or connect to one from the GUI.

rootzoll, 2019-06-23:
Thanks ... the issue list for the v1.3 is already quite long, so will push this a bit up the line .. but its most def a feature to support better in the future.

CommanderPoe, 2019-06-25:
> Got the altruistic watchtower function working on my RaspiBlitz:
> https://github.com/openoms/lightning-node-management/blob/master/watchtower.md

The process seems fairly simple. Apart from the space required (store all previous channelstates) to implement a watchtower with another peer is there any other drawback? How simple would it be to make it happen between you and I for example? What happens if we both go offline? Can i be the watchtower for a friend and viceversa? Would it need any hardware improvement? 

Maybe this are some silly questions but i still dont get it 100%. Gonna listen later to the latest SLP whose gues is Bitconner, maybe some of my answers get replied there :/ .

openoms, 2019-06-25:
> The process seems fairly simple. Apart from the space required (store all previous channelstates) to implement a watchtower with another peer is there any other drawback?

For the watchtower there will be some hardware needs (hard to estimate as dependent on the monitored channel numbers and activity):
* more diskspace and some more RAM usage
* some added computational burden
* more network traffic

I would presume that even the RPi3 can deal with monitoring a small, low traffic node, but as above it is all case dependent. More powerful hardware takes more to reach it`s limits.

> How simple would it be to make it happen between you and I for example? 

Very simple as described in the guide, you just need to know my IP address (or dynamic DNS)

> What happens if we both go offline?

If the client node is online, the watchtower only receives it\`s updated states.
The watchtower needs to be online to serve it\`s role (monitor the blockchain fro a breach).

If the watchtower is offline while the the client is on it might miss some channel state updates. I think it only monitors for possible breaches with states which have been communicated to it. If the watchtower is offline the client cannot send it`s state updates, so those will be missed and can be cheated with.

> Can i be the watchtower for a friend and viceversa? Would it need any hardware improvement?

Yes, see both above.

rootzoll, 2020-04-01:
Marked this idea for the Lightning HackSprint - details see: https://wiki.fulmo.org/index.php?title=Lightning_HackSprint

The tasks would be:

- Make a script that activates/deactivates the Watchtower-Service in LND - see notes by openoms: https://github.com/openoms/lightning-node-management/blob/master/watchtower.md
- Make it easy with that script to add/remove other LND nodes to the list of nodes it watches with a SSH GUI (dialog or whiptail)

rootzoll, 2020-04-08:
Note about automatic TOR address for watchtower service in upcoming LND:
https://github.com/lightningnetwork/lnd/blob/master/docs/watchtower.md#tor-hidden-services

rootzoll, 2020-05-27:
v1.6 will come with a SSH UI to manage "subscriptions" - it will start with payed subscriptions for IP2TOR bridges but that can also be the place to add those voluntary watchtower "subscriptions".

tlindi, 2022-12-22:
Any progress to support LND - WTClient "Subcription to connect WatchTower"? 

How about also support to config node as WT Server (listen to clearnet 9911 would work for me)?

rootzoll, 2022-12-24:
@tlindi the idea is good - it would best be a `lnd.watchtower.sh` script that uses ssh dialogs to let people config the own LND to either provide or adding a friends watch tower. Basically its just reconfiguring the LND config file. Like any good idea this needs somebody from the community doing it. If anybody wants set a bounty on it - I am happy to promote.