raspiblitz/raspiblitz #670: Can't recover funds after HDD crash

rytisbitcoin, 2019-06-24:
I had ~0.045 BTC in on-chain balance, and ~0.01 BTC in channels (2 channels, one with ~0.01 BTC local balance and one with ~0.01 BTC remote balance; I have asked the other party to close both of these channels before doing the procedure described further). I also have the backup seed, wallet.db and channel.backup files.

At first, I tried restoring as suggested in the FAQ (done a new setup of raspiblitz, and selected restore old wallet using the seed and channels.backup file). This showed 0 balance everywhere.

I then manually copied wallet.db file over, and my on-chain balance was shown as ~0.045 BTC. No channel balance however. I have listed all channels, and the two channels I had were displayed as pending close (however, they did not show any funds in limbo). I have tried to force close the channel which had my ~0.01 BTC local balance. This forced the LND node to crash and to get stuck in a permanent reboot loop.

I have reflashed the SD card and recopied wallet.db and channel.backup to the HDD, but it is still stuck in a reboot loop (I have somehow managed to make it show the on-chain balance between reboots, and it still shows ~0.045 BTC on-chain balance).

Error message shown:

****************************************
* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

17 RESTARTS DETECTED - LND might be in a error loop
Jun 24 22:31:22 raspinode lnd[4180]: main.(*server).Start.func1()
Jun 24 22:31:22 raspinode lnd[4180]:         /Users/roasbeef/gocode/src/github.com/lightningnetwork/lnd/server.go:1129 +0x324
Jun 24 22:31:23 raspinode lnd[4180]: sync.(*Once).Do(0x2a52008, 0x2890dec)
Jun 24 22:32:41 raspinode lnd[5852]: panic: runtime error: invalid memory address or nil pointer dereference

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

After using command /home/admin/XXdebugLogs.sh | nc termbin.com 9999, it generated these logs: https://termbin.com/eljd

CommanderPoe, 2019-06-25:
for off-chain funds recovery you need to replace channelstate.db

rytisbitcoin, 2019-06-27:
Where is channelstate.db located?

ashwilly, 2019-07-11:
I know someone who once helped me with my wallet recovery, you can contact run contact on cheng66 @ protonmail.ch

rootzoll, 2019-07-28:
If you have still the seed & channel.backup file (since version v1.2 of RaspiBlitz) please keep them safe. Still waiting for issue #632 getting resolved. @openoms is you can get somebody from LND finally to help on that this would be great ... having such bad experiences on recovery with the SCB is not building up trust.

Copying over manual data will often cause side effect that are not a "bug". For recovering with the channels.backup file see #632. I will close this issue - but feel free to report on your progress.

closed 2019-07-28 by rootzoll