raspiblitz/raspiblitz #673: Prepare and Test for RaspberryPi 4

rootzoll, 2019-06-27:
The RaspberryPi 4 got released and RaspiBlitz v1.3 should be ready for this great hardware update. 

TODO: Build SD card based on "Raspbian Buster with desktop"
https://www.raspberrypi.org/downloads/raspbian/ 

TODO: Test with RaspberryPi 4
TODO: Test with RaspberryPi 3 compatibility

rootzoll, 2019-06-28:
Compatibility test with RaspberryPi 3 tests looking good

rootzoll, 2019-06-29:
OK new v1.3 Raspiblitz test image build on Raspbian Buster is running fine one RaspberryPi 4 :)

rootzoll, 2019-06-29:
Powertest needs to be updated:
On RaspberryPi4 the default seems with stable power 0.8718 V / 871800 microVolt - starting form 76 C temp it seems to throttle down the CPU speed.
TODO: How to detect a RaspberryPi4?

rootzoll, 2019-06-29:
OK - Power/Stresstest can now detect RaspberryPi 4 and adapts the treshhold values.

rootzoll, 2019-06-29:
Ordered https://shop.pimoroni.com/products/fan-shim for cooling testing

rootzoll, 2019-07-01:
Fixed the "self-sync" option because RP4 gets in the range to make the job in around 2-3 - testing at the moment.

rootzoll, 2019-07-01:
Running a "self-sync" test with RP4 .. first 20% was a overnight thing (that part where the blockchain was not that full yet) but then it gets slowly. From the look of htop CPU is just 50% and memory looks just a 1/3 is used (of 2GB). So it maybe HDD slowing things down.

Will also run "self-sync" test with SSD soon. If that works out in a fair time it may make sense to measure HDD/SSD speed on bootup and then advise to self-sync or not.

rootzoll, 2019-07-02:
**Running HDD speed tests...**

```
sudo apt-get install hdparm
sudo hdparm -Tt /dev/sda
```

**RP3 + HDD:**

Timing cached reads:   1184 MB in  2.00 seconds = 592.29 MB/sec
Timing buffered disk reads:  92 MB in  3.05 seconds =  30.13 MB/sec

**RP3 + SSD:**

Timing cached reads:   1238 MB in  2.00 seconds = 619.16 MB/sec
Timing buffered disk reads:  94 MB in  3.03 seconds =  31.05 MB/sec

**RP4 + HDD:**

Timing cached reads:   1452 MB in  2.00 seconds = 726.00 MB/sec
Timing buffered disk reads: 372 MB in  3.01 seconds = 123.61 MB/sec

**RP4 + SSD:**

TODO

openoms, 2019-07-02:
Testing the RPi4 4GB with the official 3A USB-C powersupply.
The HWtest was failing, so disconnected everything (only bare board with the power supply, ethernet and SDcard attached):
```
*** Hardware Test Report ***
RaspiBlitz Hardwaretest v0.2
RaspberryPi Model Version: 4
Starting sysbench to run for 60 seconds (--max-time=60 --cpu-max-prime=10000)
58.0'C 1500 MHz  0.8630V
--> Power WARN detected
65.0'C 1500 MHz  0.8630V
--> Power WARN detected
67.0'C 1500 MHz  0.8630V
--> Power CRITICAL detected
--> Power WARN detected
68.0'C 1500 MHz  0.8630V
--> Power CRITICAL detected
--> Power WARN detected
70.0'C 1500 MHz  0.8630V
--> Power CRITICAL detected
--> Power WARN detected
--> Temp WARN detected
63.0'C  600/1500 MHz 0.8630V
--> Power CRITICAL detected
--> Power WARN detected
62.0'C  600/1500 MHz 0.8630V
--> Power CRITICAL detected
--> Power WARN detected
59.0'C 1500 MHz  0.8630V
--> Power CRITICAL detected
--> Power WARN detected
59.0'C 1500 MHz  0.8630V
--> Power CRITICAL detected
--> Power WARN detected
60.0'C  600/1500 MHz 0.8630V
--> Power CRITICAL detected
--> Power WARN detected
58.0'C  600/1500 MHz 0.8630V
--> Power CRITICAL detected
--> Power WARN detected
58.0'C  600/ 600 MHz 0.8630V
--> Power CRITICAL detected
--> Power WARN detected
```

It seems that my default voltage is 0.8630 V.
I have no heatsink on and as it looks the CPU is severely throttled above 60 degrees.

openoms, 2019-07-02:
> **Running HDD speed tests...**
>
> ```
> sudo apt-get install hdparm
> sudo hdparm -Tt /dev/sda
> ```
>
> **RP3 + HDD:**
>
> Timing cached reads: 1184 MB in 2.00 seconds = 592.29 MB/sec
> Timing buffered disk reads: [...]

**RPi4 + SSD:**  (Samsung 860 EVO 500GB)

Timing cached reads:   1706 MB in  2.00 seconds = 853.70 MB/sec
Timing buffered disk reads: 1104 MB in  3.00 seconds = 367.52 MB/sec

rootzoll, 2019-07-02:
The new shopping list will be RP4 focused - just mentioning that RPi3B+ also can be used if you alread have one.

The HardwareTest is already adapted for the RP4 in v1.3 branch.

openoms, 2019-07-02:
> The HardwareTest is already adapted for the RP4 in v1.3 branch

I am not sure why but the voltage on my Pi is slightly lower: 863000 mV vs. 871800 mV. It could be because of the 4GB vs 2GB RAM? Are you using the official RPi power supply too?

rootzoll, 2019-07-02:
I adapted the power test - I used my Macbook power supply ;)

openoms, 2019-07-03:
Running the Initial Block Download (self-sync) behind Tor on the RPi4 4GB and SSD - first 30% was ~12 hours. Already quicker than my Odroid HC1/XU4 was on clearnet.

CommanderPoe, 2019-07-03:
Really interesting tests happening here guys, hopefully, we have that shopping list updated really soon. Well done.

rootzoll, 2019-07-03:
OK testes the Shim fan for active cooling. It fits under the LCD and the LCD is still working. Its effective cooling - even in stress test the temp gets just to upper 50s. In normal operations mine is 45 celsius. I recommend this fan to keep temp in check on the RP4.

rootzoll, 2019-07-08:
Tested Self-Syncing/Validation: Was not able to finish 100% due to time ... but was able to sync up to 80% in about 3 days - over last night (half day) it was able todo 8% ... so its realistic to fully self-sync a RaspiBlitz with RP4&SSD in around 4-5 days. And the temp with the shim fan was during sync at around 47/48 celsius .. so super cool :)

openoms, 2019-07-08:
There is some fun python code to control the fan shim and built in LED. Probably should not be included by default, but worth a mention in the readme, it is great to play with:
https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-fan-shim

https://github.com/pimoroni/fanshim-python/tree/master/examples

openoms, 2019-07-09:
Two finished IBD times on the RPi4 4GB with and SSD behind Tor: 80 and 96.5 hours https://github.com/rootzoll/raspiblitz/issues/592

annaparker, 2019-07-11:
There's bug with power with Pi4, it will be fixed in the next months.

https://www.techrepublic.com/article/your-new-raspberry-pi-4-wont-power-on-usb-c-cable-problem-now-officially-confirmed/

woeisme, 2019-07-21:
Hello all, new to Raspiblitz and testing out the new RC3 for my rpi4
got as far as raspian and 
``wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.3/build_sdcard.sh  && sudo bash build_sdcard.sh

upon reboot it complains about not being able to find bitcoind
https://termbin.com/35bq

woeisme, 2019-07-21:
I noticed an issue with flashing the http://raspiblitz.com/raspiblitz-v1.3RC1-2019-07-04.img.gz to sdcard only when I did  `zcat raspiblitz-v1.3RC1-2019-07-04.img.gz | sudo dd bs=4M  of=/dev/mmcblk0 conv=fsync status=progress' ` did it manage to flash the card correctly and boot.

woeisme, 2019-08-02:
[I've created a new issue from RC3 regarding RTL hangs](https://github.com/rootzoll/raspiblitz/issues/700)

openoms, 2019-08-03:
The Initial Block Download should not be called `ultra slow` in case the acripts are running on any more powerful hw (2GB or more RAM and USB3) than the RPi3 . That is the best option to start the node and should be communicated to the user in the setup menu.

openoms, 2019-08-28:
Even lower voltage results were spotted on Telegram:
HDMI external monitor, also the touch screen attached, keyboard and mouse, HDD:  USB Seagate 1TB.
```
— WITH USB HDD attached —
67.0'C 1500 MHz 0000000000 0.8332V
66.0'C 1500 MHz 0000000000 0.8332V
66.0'C 1500 MHz 0000000000 0.8332V
65.0'C 1500 MHz 0000000000 0.8332V
65.0'C 1500 MHz 0000000000 0.8332V
64.0'C 1500 MHz 0000000000 0.8332V

 — WITHOUT USB HDD attached —
69.0'C 1500 MHz 0000000000 0.8332V
67.0'C 1500 MHz 0000000000 0.8332V
67.0'C 1500 MHz 0000000000 0.8332V
66.0'C 1500 MHz 0000000000 0.8332V
66.0'C 1500 MHz 0000000000 0.8332V
66.0'C 1500 MHz 0000000000 0.8384V
65.0'C 1500 MHz 0000000000 0.8332V
```

As attaching the HDD seems to make no difference I suggested to carry on with the blockchain download. Will follow up. We might need to reduce the power limit further if there will be no issues here.

rootzoll, 2019-08-28:
@openoms as long there are no "Bad Power" reports in the logs we should be fine to lower limit in the stress test for RP4 still a bit.

closed 2019-09-03 by rootzoll