raspiblitz/raspiblitz #674: Raspiblitz Error, no connection possible

WebBlockChainDev, 2019-06-28:
I got a problem with my raspiblitz, i cant connect through ssh and cant connect throug RTL. Also cant get access to the HDD through USB to PC. What can i do here to backup my funds / btc in open channel?

https://i.imgur.com/24qB1NO.jpg

fluidvoice, 2019-07-22:
Windows does not have the ability to access the HDD which is formated in Linux as Ext4 partition.
But you can install software that will give you access: https://www.howtogeek.com/112888/3-ways-to-access-your-linux-partitions-from-windows/

rootzoll, 2019-07-28:
This looks like a external recovery scenario. Try to connect the HDD on another compter and backup the data. 

Like fluidvoice was mentioning ... the ext4 Filesystem of the RaspiBlitz HDD is just supported by linux natively. You can try to create a LiveLinux (Ubuntu booting from a DVD or USB stick) for your laptop and then try to connect.

Because this is not a RaspiBlitz "Bug" . i will close the issue - but feels free to post further progress on your issue.

closed 2019-07-28 by rootzoll

WebBlockChainDev, 2019-08-15:
I had no time to continue yet with this problem but today i started reading about it more.

I m not sure how to recover my funds here now. 
I got 25% of my funds in the wallet and 75% in channels.

Since i have a older version (i m not sure where to see which version), i dont have the following things to recover my funds:
v1.1 lnd.rescue.sh
v1.2 channel.backup

So i m not sure now what todo without losing funds. I read alot that people r able to get their wallet funds through their Seed but they dont get the channel funds back. Some even dont get the funds of the wallet balance what makes me worry now since my amount is not that small here.