raspiblitz/raspiblitz #675: New Menu - Copy Blockchain

CryptoSigmund, 2019-07-05:
I think it is useful if one raspiblitz owner can just plug in another usb drive and go into the menu with something like "copy blockchain" then it will check for a fresh plugged in HDD and suggest to format it in ext4 and copy the blockchain from the already up to date to the new one.

So a friend can now even get faster to his final setup and don't have to download the whole blockchain. This is something like in the first sync options but now the other way around.

What you guys think about it?

Greetings!

rootzoll, 2019-07-06:
That feature already exists and its called "CLONE" during setup. What makes
it a bit complicated is that the RaspberryPi is not good at powering two
USB HDDs without additional power supply. It should work if you put a
powered USB Hub inbetween. And now with USB3 on RP4, transfer speed should
be in reasonable area.

If you like you can retest that feature on v1.3 RC1 (see v1.3
branch README for download) and let me know.

CryptoSigmund, 2019-07-07:
Generel:
Power Test failed 5 times no success with Macbook USB-C power supply. Didn't figure out why?!

2 SSDs:
HDD Clone gets started but Raspberry hangs after a few minutes

2 SSDs (1 with external power):
HDD Clone get stucked at 98% and started new with now error screen

Copy to 98% was about 5:41 but then no more response... 
[image]

Can't get the original USB-C Raspi power supply caused be the high demand. Does this work for you better?

Could you do a shorter update period on a hdd copy with the speed and % - always get paranoid staring at the screen if it will update.

rootzoll, 2019-07-28:
The USB-C power supplies are in testing. Most important is that you dont get a "undervoltage" report from them. Ma Macbook power supply also showed those - even it has a lot of amps. The original RP4-PS looks good so far but also in stress test goes quite low on the voltage - but no reports. This will be a longer process to fine tune.

TODO: Retesting the CLONE option for 1.3 release.

rootzoll, 2019-08-05:
OK - did test run with CLONE. Basics seem to work again.

closed 2019-08-05 by rootzoll

CryptoSigmund, 2019-08-07:
Thank you for testing this again. 

Can you implement that a user can choose between 

-Clone the whole disk to a new disk  (when you want to change the existing disk)
and
-Copy just the blockchain to a new disk (when a friend needs the blockchain or you want to prepare for a hackday)

all at an actual running and fully functional raspi, not in the setup routine.

rootzoll, 2019-08-07:
> Clone the whole disk to a new disk (when you want to change the existing disk)

Thats a complete new feature that does not exist yet. Please open a new issue for that and may make sense to put it in a coming release because with the RP4 people might consider also upgrading to a a SSD soon.

> Copy just the blockchain to a new disk (when a friend needs the blockchain or you want to prepare for a hackday)

This is the reverse setup as with the CLONE during setup now. With CLONE I add a friends HDD to my fresh RaspiBlitz to copy that blockchain data over. You want to connect a fresh HDD to a running RaspiBlitz to prepare it with blockchain data. There is already a similar feature called "COPYSTATION" to use one RaspiBlitz to prepare multiple new HDDs (and keep them in sync) for a HackDay/Workshop setup - see the WORKSHOP.md --> https://github.com/rootzoll/raspiblitz/blob/master/WORKSHOP.md#prepare-blockchain-copy-station But that feature is specialised to prepare multiple new HDDs so it will on the start first make a local template folder and duplicate its own blockchain data to have one version to further sync with the network (moving data), while copying the other folder (static data) to the connected HDDs (on a powered USB Hub) - that way you can prepare 10 HDDs for a workshop at once ... use ud RP4 as a copystation is recommended because of USB3 speedup on copy.