raspiblitz/raspiblitz #677: Pending force closing channels not closing / limbo balance

j-ockeghem, 2019-07-07:
I'm running RaspiBlitz 1.2 and I wanted to shut down my node. However, several channels are shown as "Pending Force Closing" for several weeks. "Block till maturity" is actually negative. For some, there is a "limbo balance" of the same amount as the "local balance" shown. Screenshot:

[image]

I had closed the channels with RaspiBlitz' option "Close all open channels". 

Is there anything I can do to get my funds back, or is this a kind of bug and my funds are lost? I googled already and found several users with the same problem, but did not find a solution. Any help would be appreciated.

j-ockeghem, 2019-07-07:
Update: It seems the channels are slowly closing. Now I am down from 22 to 7 "pending force closing channels". I hope they will also close. I had initiated the "close all open channels" 2-3 weeks ago, but had then shut down my Raspiblitz. When I turned it back on, I waited for bitcoind and lnd to be fully sync'ed, but the channels still showed as pending force close. There seems to be some lag-time and I don't know where it's coming from. In any case, I find it very frightening to think some of my BTC my be locked in channels forever and lost. I hope lnd (or another client) can get more user-friendly in that respect.

th4fl4sh4, 2019-07-07:
I have the same problem. One of the pending force close channels is now pending for over 2 months. 
The status shows: Blocks Til Maturity: -8061 
There is clearly something wrong here. Any help would be highly appreciated.

j-ockeghem, 2019-07-08:
@th4fl4sh4 2 months, oh my god. I am now down to 5 "pending force closing" channels. Tried rebooting the RaspiBlitz as well, but nothing changed in the past 24h:

[image]

I still hope @rootzoll or someone else might have a suggestion of what to do to unlock these funds.

If these cases are not handled well, I assume more and more people will dismiss lightning network. I am definitely taking a break after this frustrating experience.

openoms, 2019-07-08:
Check the channel from the command line with `lncli listchannels` and see if that matches the info in RTL.

osilloh, 2019-08-22:
I had an issue with a stuck channel with LNBIG.com as well. Oddly, when I check 1ml.com it is still seen as active for over a month..

I tried rebooting (Pi3 running 1.2) and now I am stuck in a loop. This is the farthest it will get, before I get a kernel panic.

[image]

[image]

 I have attached my debug output. Any help would be appreciated. https://termbin.com/mu0y

rootzoll, 2019-08-28:
@osilloh most of those kernel panic errors should be fixed with the updated lnd in v1.3 .. that is not final yem but you can find a almost close to final Release Candidate image in the v1.3 branch:
https://github.com/rootzoll/raspiblitz/tree/v1.3#installing-the-software

frennkie, 2020-07-23:
Is there a clearly defined issue/request in here which we from RaspiBlitz side can/should fix/implement? 

If not I would close the issue.

rootzoll, 2020-07-23:
The kernel panics were due to an old LND version. I will close this issue.

closed 2020-07-23 by rootzoll