raspiblitz/raspiblitz #678: RaspiBlitz v1.3RC1 - nodeJs install failure when installing RTL

openoms, 2019-07-08:
error is:
```
admin@RPi4:~ $ curl -sL https://deb.nodesource.com/setup_11.x | sudo -E bash -

## Installing the NodeSource Node.js 11.x repo...

## Populating apt-get cache...

+ apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]                           
Hit:3 https://deb.torproject.org/torproject.org stretch InRelease                                   
Reading package lists... Done                       
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:7
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:7
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
Error executing command, exiting
```

Will need to download the compressed package instead as per: https://github.com/nodejs/help/wiki/Installation

openoms, 2019-07-08:
Testing https://github.com/ShahanaFarooqui/RTL/releases/tag/v0.4.2 now.

openoms, 2019-07-11:
issue confirmed with another tester. RTL was running ok when installed manually.

rootzoll, 2019-07-28:
TODO: verify binaries: https://github.com/nodejs/node#verifying-binaries

fluidvoice, 2019-07-28:
> issue confirmed with another tester. RTL was running ok when installed manually.

I also could not install RTL via my VM install of RC1.

openoms, 2019-07-28:
the fix has been merged now https://github.com/rootzoll/raspiblitz/pull/679
Pull the updated installer script:
`wget https://raw.githubusercontent.com/rootzoll/raspiblitz/master/home/admin/config.scripts/bonus.rtl.sh`

rootzoll, 2019-07-29:
OK RTL is fixed by PR of @openoms and I added checksum validation of the downloaded NodeJS binaries. Will be partof final v1.3 release. Closing issue.

closed 2019-07-29 by rootzoll