raspiblitz/raspiblitz #680: extracted img file hash

rafaelalvespereira, 2019-07-09:
Hi! I extracted the .gz file using Linux and got a 4Gb file. Then I extracted it on Windows (7-zip) and got a 8Gb file. Which one is correct? What is the SHA256 of the .img file?

Dimpelmoser, 2019-07-09:
... the SHA256 for checking the file <raspiblitz-v1.2-2019-05-02.img.gz> after the download can be found on https://github.com/rootzoll/raspiblitz below the download-links.
To bring the image file to the SD card it may be helpful to read https://www.raspberrypi.org/documentation/installation/installing-images/README.md.

rafaelalvespereira, 2019-07-10:
Thanks. I checked the SHA256 for the .gz file and it is correct. But when extracted, on Linux and on Windows, I got two different .img files, with different hashes.

openoms, 2019-07-11:
> Thanks. I checked the SHA256 for the .gz file and it is correct. But when extracted, on Linux and on Windows, I got two different .img files, with different hashes.

There is no need to extract, just burn the compressed img.gz directly with Etcher.

woeisme, 2019-07-22:
I noticed an issue with flashing the http://raspiblitz.com/raspiblitz-v1.3RC1-2019-07-04.img.gz to sdcard only when I did:

zcat raspiblitz-v1.3RC1-2019-07-04.img.gz | sudo dd bs=4M of=/dev/mmcblk0 conv=fsync status=progress'

did it manage to flash the card correctly and boot.

perhaps this is related to [**Section** Copying a zipped image to the SD card](https://www.raspberrypi.org/documentation/installation/installing-images/linux.md)

rootzoll, 2019-07-28:
TODO: Add info to v1.3 README on writing sd card linux

closed 2019-09-04 by rootzoll