raspiblitz/raspiblitz #681: New Menu Item - Rotate Screen

CryptoCloaks, 2019-07-16:
Hello RaspiBlitzers, we were wondering if it was possible to add a menu item to allow users to rotate the screen 180 degrees to allow better cord management when wall mounting a unit.

Dimpelmoser, 2019-07-18:
@CryptoCloaks 
It's not a menu item, but maybe that helps too ... #491

openoms, 2019-07-22:
> Hello RaspiBlitzers, we were wondering if it was possible to add a menu item to allow users to rotate the screen 180 degrees to allow better cord management when wall mounting a unit.

thanks @Dimpelmoser . This is the manual option: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#can-i-flip-the-screen
Not sure if it used that often that it would need a menu option, but would be an easy PR :) .

rootzoll, 2019-07-28:
OK I added a rotate config script that can be set from the services menu to flip the lcd 180 degree. This will be part of the v1.3 final release. Closing Issue.

CryptoCloaks, 2019-07-29:
Thanks everyone! This will be awesome to have for RaspiBlitzer's and wall mounting!

Himbeergeld, 2020-02-22:
After updating to V1.4 (coming from V1.3 on my Pi3) the screen was flipped. It seems the default rotation value has changed. I tried to change the rotation back with the Rotate Screen menu but it had no effect.
Doing sudo nano /boot/config.txt I realized that there are actually TWO dtoverlay variables containing the rotation settings.
Only changing the SECOND dtoverlay variable to 270 actually had an effect: 
dtoverlay=tft35a:rotate=270
Is it possible that the "Rotate Screen" menu only effects the first dtoverlay variable?

openoms, 2020-02-22:
@Himbeergeld 
There should be only one entry so this is a bug.
The menu setting should change both.
The solution is covered in the FAQ: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-fix-my-upside-down-lcd-after-update
but we can improve for the next release.

rootzoll, 2020-03-21:
OK tested for v1.5 release .. on updating from a v1.3 it now keeps the screen rotation to the old default. Closing issue.

closed 2020-03-21 by rootzoll