raspiblitz/raspiblitz #682: Will Waveshare 3.2 inch LCD 320X240 TFT Display work? It fits inside the suggested case.

LukeParker, 2019-07-19:
I bought all of the suggested (USA) parts except for the LCD screen, because many have reported that the suggested case doesn't close over the top of that one, but it will over this one: 
https://www.amazon.com/gp/product/B01HPV68UM/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1

I ordered it and assembled it, and it does fit like a glove, but the drivers in your image must be for the wrong screen... On boot it starts with a fully white screen and never changes, despite the HDMI's monitor working normally.

I've got Waveshare's drivers but no idea how to put them into your image, nor any other way to continue. Help!

openoms, 2019-07-22:
It should work. Not sure which is the right driver form the Amazon link, but you can try until works. Just follow the tutorial given on their website: https://www.waveshare.com/wiki/3.2inch_RPi_LCD_(C)

LukeParker, 2019-07-23:
Thanks! Got it to work.
When I went to do exactly as written above I got a Fatal exception  because it said it would write over the entire LCD-show folder and it wouldn't let me continue.

Luckily, one of the included drivers in that folder is /LCD32-show (without the C) I took a gamble and ran it; it seems to work great, although the edges dont fill the screen as much as they probably should. I'll update that later.

closed 2019-07-23 by LukeParker