raspiblitz/raspiblitz #686: The LND service is not running

Geranni, 2019-07-24:
Had my Raspiblitz ("Melina") running a few months on 1.1. After update to 1.2, it started working very unreliably. Periods of 1-2 weeks without malfunction were interrupted by errors and now it stopped working completely. It shows the following error:

****************************************
* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

11 RESTARTS DETECTED - LND might be in a error loop
Jul 21 14:16:20 Melina lnd[17323]: github.com/coreos/bbolt.(*node).spill(0x2ebc840, 0x24753a0, 0x17)
Jul 21 14:16:20 Melina lnd[17323]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/node.go:378 +0x2f4
Jul 21 14:16:20 Melina lnd[17323]: github.com/coreos/bbolt.(*Bucket).spill(0x26922a0, 0x2692200, 0x2475570)
Jul 21 14:16:20 Melina lnd[17323]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/bucket.go:568 +0x360
Jul 21 14:16:20 Melina lnd[17323]: github.com/coreos/bbolt.(*Bucket).spill(0x248828c, 0xbf453749, 0x894a2ddf)
Jul 21 14:16:20 Melina lnd[17323]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/bucket.go:535 +0x2d8
Jul 21 14:16:20 Melina lnd[17323]: github.com/coreos/bbolt.(*Tx).Commit(0x2488280, 0x0, 0x0)
Jul 21 14:16:20 Melina lnd[17323]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/tx.go:160 +0x90
Jul 21 14:16:20 Melina lnd[17323]: github.com/coreos/bbolt.(*DB).Update(0x27f8500, 0x2475678, 0x0, 0x0)
Jul 21 14:16:20 Melina lnd[17323]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/db.go:695 +0xd4
Jul 21 14:16:20 Melina lnd[17323]: github.com/lightningnetwork/lnd/channeldb.(*ChannelGraph).SetSourceNode(0x24817d0, 0x2488200, 0x46, 0xc88e715c)

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

After using the /home/admin/XXdebugLogs.sh command I get the following output:

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 14:18:00 up 31 min,  2 users,  load average: 2.99, 2.95, 2.64

*** BCLOCKHAIN SYSTEMD STATUS ***
? bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-07-21 13:49:11 BST; 28min ago
  Process: 2236 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 2228 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 2295 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           ??2295 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Jul 21 13:49:10 Melina systemd[1]: Starting Bitcoin daemon...
Jul 21 13:49:11 Melina systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sat 2019-07-20 19:57:57 BST, end at Sun 2019-07-21 14:18:00 BST. --
Jul 21 13:49:10 Melina systemd[1]: Starting Bitcoin daemon...
Jul 21 13:49:11 Melina systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-07-21T13:13:36Z UpdateTip: new best=0000000000000000000a50b1c6dbc5db6b2e68f0450eee0ce39eb64688ec084a height=586295 version=0x20000000 log2_work=90.868912 tx=437034443 date='2019-07-21T00:04:00Z' progress=0.999740 cache=16.7MiB(148279txo) warning='39 of last 100 blocks have unexpected version'
2019-07-21T13:13:39Z UpdateTip: new best=00000000000000000007ccbe3aa8fc2c26aa3c8dd2072930bd90eb5bab6f43ef height=586296 version=0x20000000 log2_work=90.868937 tx=437035120 date='2019-07-21T00:06:47Z' progress=0.999741 cache=16.8MiB(148962txo) warning='39 of last 100 blocks have unexpected version'
2019-07-21T13:13:41Z UpdateTip: new best=0000000000000000001414b4e32ab3c89917bf9309a8036a903c899944b7f7bc height=586297 version=0x20000000 log2_work=90.868961 tx=437035744 date='2019-07-21T00:08:13Z' progress=0.999741 cache=16.9MiB(149383txo) warning='39 of last 100 blocks have unexpected version'
2019-07-21T13:13:51Z UpdateTip: new best=0000000000000000000b9afa69cf58918ee1edac8cc1126edad70d840995f32a height=586298 version=0x20000000 log2_work=90.868986 tx=437037204 date='2019-07-21T00:14:55Z' progress=0.999743 cache=17.1MiB(151494txo) warning='39 of last 100 blocks have unexpected version'
2019-07-21T13:14:33Z UpdateTip: new best=0000000000000000000f5509d0c19df310bf3ed3d55c83b44581a34bb62ef2b5 height=586299 version=0x20000000 log2_work=90.869011 tx=437040172 date='2019-07-21T00:49:58Z' progress=0.999755 cache=17.7MiB(157473txo) warning='39 of last 100 blocks have unexpected version'
2019-07-21T13:14:58Z UpdateTip: new best=0000000000000000000fd6d7eda96145bb2b4978e416db1122a37410a7ac7d12 height=586300 version=0x20c00000 log2_work=90.869036 tx=437043136 date='2019-07-21T00:50:54Z' progress=0.999755 cache=18.2MiB(161909txo) warning='40 of last 100 blocks have unexpected version'
2019-07-21T13:15:28Z UpdateTip: new best=00000000000000000009cc98872e3c33d0d67614bb454df21b9961c524f2c655 height=586301 version=0x20000000 log2_work=90.869061 tx=437046003 date='2019-07-21T01:03:20Z' progress=0.999759 cache=18.7MiB(166834txo) warning='39 of last 100 blocks have unexpected version'
2019-07-21T13:15:31Z UpdateTip: new best=000000000000000000142c167c4bb65f41238bb3a503e1d8daf0d4b7513366e2 height=586302 version=0x3fffe000 log2_work=90.869086 tx=437046695 date='2019-07-21T01:03:48Z' progress=0.999759 cache=18.8MiB(167183txo) warning='39 of last 100 blocks have unexpected version'
2019-07-21T13:15:36Z UpdateTip: new best=0000000000000000000d31af6e20c0465ac1cfeaf5899f7d86f0cc78eef40b6d height=586303 version=0x20000000 log2_work=90.869111 tx=437047786 date='2019-07-21T01:07:28Z' progress=0.999760 cache=18.9MiB(168461txo) warning='38 of last 100 blocks have unexpected version'
2019-07-21T13:15:39Z Pre-allocating up to position 0x900000 in rev01728.dat
2019-07-21T13:15:39Z UpdateTip: new best=00000000000000000019945271d02c2e9cdcf50f024fd524a4f973fcea4372ba height=586304 version=0x20000000 log2_work=90.869135 tx=437048426 date='2019-07-21T01:09:55Z' progress=0.999761 cache=19.0MiB(169064txo) warning='37 of last 100 blocks have unexpected version'
2019-07-21T13:15:49Z UpdateTip: new best=00000000000000000015031f6abaa900bff18dad6484e97e09774bea65ad7dc3 height=586305 version=0x3fff0000 log2_work=90.86916 tx=437050496 date='2019-07-21T01:18:26Z' progress=0.999764 cache=19.2MiB(171225txo) warning='38 of last 100 blocks have unexpected version'
2019-07-21T13:16:15Z UpdateTip: new best=00000000000000000009a4a9165c8e6f34652497574b0f061fabea1dced2d9b9 height=586306 version=0x20000000 log2_work=90.869185 tx=437052293 date='2019-07-21T01:26:55Z' progress=0.999766 cache=19.6MiB(175150txo) warning='37 of last 100 blocks have unexpected version'
2019-07-21T13:16:18Z UpdateTip: new best=000000000000000000180bc2d25df0b48ed80884acef009dde6d00820322837d height=586307 version=0x20800000 log2_work=90.86921 tx=437052958 date='2019-07-21T01:30:05Z' progress=0.999767 cache=20.3MiB(181279txo) warning='38 of last 100 blocks have unexpected version'
2019-07-21T13:16:54Z UpdateTip: new best=0000000000000000001e0ff90c966ea7d1792de28f94fcc91d5b293852bc89c5 height=586308 version=0x20000000 log2_work=90.869235 tx=437054670 date='2019-07-21T01:38:36Z' progress=0.999770 cache=20.5MiB(182974txo) warning='37 of last 100 blocks have unexpected version'
2019-07-21T13:16:55Z Pre-allocating up to position 0x8000000 in blk01728.dat
[... 66 lines cut ...]
*** LAST 20 RTL LOGS ***
-- Logs begin at Sat 2019-07-20 19:57:57 BST, end at Sun 2019-07-21 14:18:00 BST. --
Jul 21 13:49:11 Melina systemd[1]: Started RTL daemon.

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
powerFAIL=0
powerWARN=1
powerMIN='1200000 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='4290 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.10.33'
tempCelsius='65.5'
uptime=1878
startcountBlockchain=1
bitcoinActive=1

It would be great if you could help me fix the error so I can get my node up and running again.

PS: There ist some misspelling in the word "Blockchain" as it reads "BCLOCKHAIN" in your code ;-)

CandleHater, 2019-07-25:
Your uploaded log shows "progress=0.999741", which means the Bitcoin Blockchain is not fully up to date. Please wait until this progress is 100%, afterwards LND will start up again and you should be good to go. 

Regarding the Blockchain typo I've created a pull request https://github.com/rootzoll/raspiblitz/pull/687, thank you for the info.

Geranni, 2019-07-27:
Despite reaching 100% sync it is still not running:

****************************************
* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

9 RESTARTS DETECTED - LND might be in a error loop
Jul 27 11:39:35 Melina lnd[13315]: panic: assertion failed: write: circular dependency occurred
Jul 27 11:40:57 Melina lnd[14785]: panic: assertion failed: write: circular dependency occurred

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 11:41:39 up 14 min,  2 users,  load average: 0.85, 0.89, 0.73

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-07-27 11:29:38 BST; 12min ago
  Process: 4103 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 4096 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 4162 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─4162 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Jul 27 11:29:37 Melina systemd[1]: Starting Bitcoin daemon...
Jul 27 11:29:38 Melina systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Sat 2019-07-27 11:41:39 BST. --
Jul 27 11:29:37 Melina systemd[1]: Starting Bitcoin daemon...
Jul 27 11:29:38 Melina systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-07-27T10:31:19Z dnsseed thread start
2019-07-27T10:31:19Z addcon thread start
2019-07-27T10:31:19Z opencon thread start
2019-07-27T10:31:19Z init message: Done loading
2019-07-27T10:31:19Z msghand thread start
2019-07-27T10:31:30Z Loading addresses from DNS seeds (could take a while)
2019-07-27T10:31:46Z 217 addresses found from DNS seeds
2019-07-27T10:31:46Z dnsseed thread exit
2019-07-27T10:32:15Z New outbound peer connected: version: 70015, blocks=587266, peer=0
2019-07-27T10:32:23Z UpdateTip: new best=0000000000000000000cb8612d53585cfe3d194a9b1670e7b23f4e1c6d9cfd0c height=587265 version=0x20000000 log2_work=90.892727 tx=439026572 date='2019-07-27T10:27:01Z' progress=0.999998 cache=1.6MiB(14310txo) warning='38 of last 100 blocks have unexpected version'
2019-07-27T10:32:43Z UpdateTip: new best=000000000000000000131f791d9897e30e43d60847d8c469a9c1bf17aac14690 height=587266 version=0x20000000 log2_work=90.892752 tx=439028754 date='2019-07-27T10:29:20Z' progress=0.999999 cache=2.3MiB(20361txo) warning='38 of last 100 blocks have unexpected version'
2019-07-27T10:32:43Z New outbound peer connected: version: 70015, blocks=587266, peer=2
2019-07-27T10:32:53Z New outbound peer connected: version: 70015, blocks=587266, peer=4
2019-07-27T10:33:00Z Imported mempool transactions from disk: 3726 succeeded, 1273 failed, 0 expired, 0 already there
2019-07-27T10:33:15Z New outbound peer connected: version: 70015, blocks=587266, peer=5
2019-07-27T10:33:15Z New outbound peer connected: version: 70015, blocks=587266, peer=6
2019-07-27T10:33:21Z New outbound peer connected: version: 70015, blocks=587264, peer=7
2019-07-27T10:33:26Z New outbound peer connected: version: 70015, blocks=587266, peer=8
2019-07-27T10:33:33Z New outbound peer connected: version: 70015, blocks=587266, peer=9
2019-07-27T10:37:56Z UpdateTip: new best=00000000000000000017724f56f03a80c79fd7fb03898e9f1ef89cbb8583a51c height=587267 version=0x20000000 log2_work=90.892776 tx=439031000 date='2019-07-27T10:37:49Z' progress=1.000000 cache=3.2MiB(27627txo) warning='38 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2019-07-27 11:40:57 BST; 42s ago
  Process: 14785 ExecStart=/usr/local/bin/lnd --tor.active --tor.v2 --listen=127.0.0.1:9735 (code=exited, status=2)
  Process: 14782 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
[... 55 lines cut ...]
Jul 27 11:29:38 Melina systemd[1]: Started RTL daemon.

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
powerFAIL=0
powerWARN=15
powerMIN='1243800 microVolt'
tempFAIL=12
tempWARN=15
tempMAX='7520 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.10.33'
tempCelsius='68.7'
uptime=872
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=587267
initialSync=0
syncProgress=100.00
startcountLightning=9
lndActive=0
scriptRuntime=1

rootzoll, 2019-07-28:
If you run v1.2 the lnd thats a 0.6.x LND ... it has recenttly an update to 0.7 with a lot of bug fixes. Those panic errors like in your debug files could be part of those fixes.

There is already a v1.3 release candidate 1 of RaspiBlitz that has the lastest LND version ... I would recommend wo wait for final release (should be beginning of august).

In the meanhile try to get a better power supply ... the hardware test shown 15 warnings. power failures can lead to data corruption LND then has problems with.

Geranni, 2019-07-29:
Thank you very much for the infos, Christian. I am looking forward to the update.

Probably an update of the shopping cart would make sense to avoid people buying inappropriate hardware (https://github.com/rootzoll/raspiblitz/blob/master/shoppinglist_de.md)
Currently I am using a 15 W power supply, do you think 20 W will be sufficient?

P-Verbrugge, 2019-08-29:
Same issue here. Setup a new SD card with v1.3. Left the HDD as it is. Still the same error. 

Do I have to backup lnd and do a new setup to be able to access my funds? 
Can’t access none of the channels now.

rootzoll, 2019-09-01:
@P-Verbrugge If you still have that error a LND rescue backup and moving it to a fresh RaspiBlitz might not fix this if the LND databases have a panic state or corruption. The hard way would be to run a setup a fresh RaspiBlitz v1.3 (final release now hopefully end of next week) and recover old wallet by seed AND channel.backup file. This will force close all active channels in the process but then builds fresh databases without that panic/corruption.

P-Verbrugge, 2019-09-01:
@rootzoll Thanks! I will wait for the the new release and give it a try. :-)

PapaCoderNit, 2019-09-03:
Im having same issue.. will wait also

rootzoll, 2019-09-09:
@PapaCoderNit the lnd-rescue tar file can be used during setup of ne fresh node to recover old funds and channels.

closed 2019-09-09 by rootzoll