raspiblitz/raspiblitz #691: RaspberryPi 4 Shopping List Updates

rootzoll, 2019-07-28:
With RaspberryPi 4 (supported by v13 of RaspiBlitz) the need for new shopping lists comes - on this issue we can collect items and then put it out as new setups.

rootzoll, 2019-07-28:
A SATA to USB adpter recommended by openoms:
https://www.ebay.co.uk/itm/254260703188?ul_noapp=true

rootzoll, 2019-07-28:
The open 3D casing build by cryptocloaks:
https://twitter.com/CryptoCloaks/status/1154412490492715009

openoms, 2019-07-30:
The Pimoroni Shim Fan: https://shop.pimoroni.com/products/fan-shim

openoms, 2019-08-06:
> A SATA to USB adpter recommended by openoms:
> https://www.ebay.co.uk/itm/254260703188?ul_noapp=true

Please pull this from the list, it does not seems to stand the test of this USB Attached SCSI (UAS) driver issue: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931
I think that is the reason why my node hangs and never did when I used the X820  board instead: https://geekworm.com/products/x820-v3-0-usb-3-0-2-5-inch-sata-hdd-ssd-storage-expansion-board (just to be clear on this the USB port is not aligned with the RPi4, so not good solution.

There is a new X825, but rather pricy 32.99 USD (+import tax and postage): https://geekworm.com/collections/raspberry-pi-4/products/raspberry-pi-4-model-b-x825-2-5-inch-sata-hdd-ssd-expansion-board

Desperate for a suitable SATA-USB adapter for the RPi4.

rootzoll, 2019-08-06:
Follow in this topic also on twitter if there additional info comes up: https://twitter.com/openoms/status/1158859796247961600

snieders, 2019-08-08:
The 'Development Board Enclosure' from MULTICOMP PRO (https://nl.farnell.com/multicomp-pro/asm-1900147-61/rpi-touch-screen-case-red/dp/3106155) in combination with the original Raspberry Pi 7" touchscreen seems like a very nice setup.

rootzoll, 2019-08-08:
@snieders the 7" touch would be another screen form factor which the RaspiBlitz is not ready for ... maybe in the future with the graphical touchscreen mode. But its a nice setup :)

openoms, 2019-08-12:
SSD adapters/enclosures confirmed working with the RPi4:

https://www.amazon.de/gp/product/B06XWSDGP6
https://www.amazon.de/gp/product/B00EEDF1Z8
(https://twitter.com/rootzoll/status/1158877054638665730)

https://www.amazon.de/dp/B00IJNDBM4/
(https://twitter.com/bavarianledger/status/1158961431729102848)

From the Telegram channel:
https://www.amazon.de/gp/product/B01GDZACDK

https://www.amazon.de/Samsung-MU-PA1T0B-EU-Portable-Kabel/dp/B074M774TW/
(https://twitter.com/wiz/status/1156722461385166848)

openoms, 2019-08-13:
The Pi Hut has better availability of the 4GB RPi4 model than pimoroni.com.
Could include in the list for the Maximal package: https://thepihut.com/products/raspberry-pi-4-model-b?variant=20064052740158

CryptoSigmund, 2019-08-13:
Considder the Shim Fan or the Aluminium Heatsink Case also for the list. 

Actual testing the heatsink case now. But for the IBD it also gets 82°C... :/

openoms, 2019-08-13:
@CryptoSigmund the shim fan is on the list already. The heatsink case is insufficienct if it gets to 82°. It does not make it above 60° with the fan.

rootzoll, 2019-08-28:
Consider replacing Power Supply with this one:
https://github.com/rootzoll/raspiblitz/issues/727

f-daniel, 2019-10-16:
Hey,

for the Switzerland shopping [list](https://github.com/rootzoll/raspiblitz/blob/master/shoppinglist_ch.md) you could add [Pi-Shop.ch](https://www.pi-shop.ch), got my Rasberry Pi 4 there as well as a fairly priced 32GB MicroSD card from SanDisk.

rootzoll, 2019-11-20:
@f-daniel thanks ... adding this issue to the v1.4 release to update the Switzerland shopping list.

0xjacobb, 2019-11-25:
> @f-daniel thanks ... adding this issue to the v1.4 release to update the Switzerland shopping list.

I made a complete change to the "shoppinglist_ch.md" based on the pi-shop.ch on my local machine. So if you want, I can create a pull request for that. I've just replaced the display with a version, which is easier to order in CH and recommended from the pi-shop.ch.

rootzoll, 2019-11-25:
@0xjacobb sure - please submit the pull request :)

closed 2019-11-25 by rootzoll

danhoule, 2020-01-06:
Re: Canada shopping list

I had all kinds of problems getting Raspiblitz/rpi configured...several times starting from scratch on SD and SSD.  Turns out that an inline pi switch (in my circumstance CanaKit USB-C PiSwitch On/Off Power Switch for Raspberry Pi 4) was causing all kinds of power related issues on the SDD resulting in input/output errors, problems ssh to device...  

Once the switch ws removed, everything worked fine.  I didn't think that the SSD would have been hyper sensitive to power supply. FWIW, be mindful of this when buying kits.

rootzoll, 2020-01-06:
@danhoule I updated the Canadian Shoppoing list (now most of the parts are available from canadian amazon anyway). I replaced the CanaKit with the a single Pi4 and the org raspberrypi power supply - see: https://github.com/rootzoll/raspiblitz/blob/master/shoppinglist_ca.md