raspiblitz/raspiblitz #692: Where is the on-chain wallet?

LukeParker, 2019-07-29:
I finally get my raspiblitz node working, (v1.2) load some sats into the on-chain wallet, enable RTL, and turn on the autobalancer. It promptly throws 50% of my sats into LND, and I assume the other 50% are still in a default bitcoin node wallet.

But, um, where is that? I seem to have no control over on-chain funds here and nothing in the node's main menu nor in RTL have any wallet controls besides lightning controls. How do I manage my on-chian funds?

rootzoll, 2019-07-29:
Its a auto-pilot (just opening channels for you with up to 50% of your funds) - its not balacing your channels - hats a feature script for the future.

Also LND is managing your on-chain funds. The bitcoind is just a information source about the blockchain.

What is the "lnblances" point in the mainmenu saying? Also in RTL unter "LND wallet" there you should see your on-chain funds.

LukeParker, 2019-07-30:
So there's no way to send to someone On-chain from my node if I happen to have money there at the time?
I did make the mistake of thinking that my LND balance was all in channels, but I don't understand where the other half of my funds are... RTL is only reporting the Local balance, not the on-chain balance.
Sadly, when I select 'lnbalance' or 'lnchannels' from the main menu (ssh) I receive the error message: 

/home/admin/00mainMenu.sh: line 109: lnbalance: command not found
Press ENTER to return to main menu.

openoms, 2019-07-30:
RTL shows the onchain balance of lnd in the `LND Wallet` tab. From there you can send and receive onchain.

The lnbalance and lnchannels bug is fixed cor the new version.

LukeParker, 2019-08-04:
Alright, it took me a little while to wrap my head around it; missing the lnbalance menu option made that more confusing then it needed to be until I finally could see my sats moving from the default wallet to channels inside RTL. 

I'd recommend putting more seperation between the two in future versions of the UI. Closing ticket, thanks to both of you for answering.

closed 2019-08-04 by LukeParker