raspiblitz/raspiblitz #697: Don't silently rewrite system config files

Kixunil, 2019-08-01:
We're conducting a code review of `build_sdcard.sh` and we've found that the script modifies dhcpd file without any notice nor interaction. This was introduced in #322.

While this may improve compatibility for some misconfigurations, it could also break things. It'd be better to have the option to turn this feature on only when necessary. The script also relies on the comment being present, so it's not very robust either.

rootzoll, 2019-08-05:
Thanks very much for the code review. Added to look into this for v1.3 release.

rootzoll, 2019-08-11:
I changed it to a more robust way to change the dns. First commenting out if sich a value was set before and the add the new config as new line at the end.

I decided to let that rewrite in there for now, because a working default is important for the rest to work - but added a comment that this can be commented out if thats preferred.

closed 2019-08-11 by rootzoll