raspiblitz/raspiblitz #698: Present configuration options before running build_sdcard.sh

Kixunil, 2019-08-01:
The script currently make lot of assumptions and defaults for the user. This might lead to undesired configuration (Including default passwords!). It should be possible to configure the parameters beforehand.

rootzoll, 2019-08-05:
Like the Raspbian sd card image also the `build_sdcard.sh` script will just create a "general image" (not a personalized one) - thats why there are still default passwords. The personal settings (including setting passwords) are done when you first run the builded image or (if you just build the sd card for yourself) after the next/final reboot.

I saw the hint to check on the dhcpd settings .. any other assumptions that might be critical? The idea of this script is to prepare the system with the most basic software/settings to run on a raspberrypi like environment. Any special extensions (like nodeJS, etc) will be installed only if the user activates the features later on. Most assumptions/defaults are based on experience like the original RaspiBolt Tutorial - see: https://stadicus.github.io/RaspiBolt/raspibolt_20_pi.html .. but of course there is still room to improve and give options. Let us know if you can see too broad assumptions/defaults that should get changed.

Kixunil, 2019-08-05:
Ah, I see. Shouldn't it only install Tor + whatever is needed for managing? The script installs tons of stuff including Bitcoind, Lnd, Litecoin (this should be definitely optional)...

I was thinking about these options:

* Features - which software should be installed - only install what the user wants
* Usernames - forcing the user to use "pi" as username isn't very friendly
* WiFi country (could be US at first and changed later and I personally don't care, but some people might be worried about legality)
* DNS fallback, as was mentioned in the other issue

rootzoll, 2019-08-05:
Bitcoind & LND are essential to have on the RaspiBlitz - its a Lightning/Bitcoin-Fullnode on a RaspberryPi ;)

> Features - which software should be installed - only install what the user wants

Basically thats already true for most more advanced features (like RTL or lndconnect). At the moment the rule of thumb is: Prepare the basics during SD card build so that a user can start with a image (for example during a workshop) without too much of install/download time to setup the basic node. Some comfort installs are OK (cli-tools, basic unix tools), when from official debian repos. Everything outside of debian repos (binaries, code from github, etc) should be reduced to minimal .. but bitcoin & lnd are essential for the idea of the node.

> Usernames - forcing the user to use "pi" as username isn't very friendly

Being based on the original tutorial the user names are essential to keep scripts consistent. Changeing them would add a lot of variables to the scripts that would add more complexity and possible errors. "pi" for example is for the LCD process. "admin" the one to work in and "bitcoin" runs the deamons (without sudo rights). If you choose a good secure password, I dont see a big downside of having a static usernames for the node.

> WiFi country (could be US at first and changed later and I personally don't care, but some people might be worried about legality)

By default the RaspiBlitz should be connected by Ethernet LAN cable - not by wifi (to have a stabe connection to the router and prevent running with bad reception, low bandwidth). The WIFI is not completly disabled - so that users can still set wifi on for example thru: `/boot/wpa_supplicant.conf` and will set country there. 

> DNS fallback, as was mentioned in the other issue

Will take a look at that issue. Thanks.

Kixunil, 2019-08-09:
> Bitcoind & LND are essential to have on the RaspiBlitz - its a Lightning/Bitcoin-Fullnode on a RaspberryPi ;)

Heh, Bitcoin of course, LND might be replaced with other impl one day in the future, but I'd agree to solve other things first.

So my understanding is that the user gets your image and attends a workshop. That actually sounds very reasonable. Considering that it creates requirement to trust binary blob from SD card creator, I'd like to see a script that generates reasonable defaults for the workshop. Then after the workshop, the user wipes the SD card and installs using the script. The script is easy to verify.

This helps even experienced devs/admins because they don't have to do it manually, but they can still check the validity. Noobs can ask a few independent friends to do the review for them.

Considering this, I'd suggest to keep the script minimal. Installing from Debian repos is a good argument, it shouldn't be malicious. Anyway, I'd suggest to install in one long command line (can be broken via variable), as that is more efficient (triggers get called only once for instance).

rootzoll, 2019-08-11:
> Anyway, I'd suggest to install in one long command line (can be broken via variable), as that is more efficient (triggers get called only once for instance).

Alone the GPG checking of bitcoind and lnd is multiple steps - that would result in a very very long and almost cryptic to check command line. To have one script to download, check and execute - like build sd card script now is much more friendly to peer review before you build your card.

I think the  build script can use some cleanup to be more simple, but it is already very close to what you expect I think.

Kixunil, 2019-08-18:
Damn, I made that phrase totally confusing! I mean `apt-get install a b` is more efficient than `apt-get install a; apt-get install b`

If you want to split it up, you can do this:

```bash
# packages related to X
INSTALL_PKGS=some_package

# packages related to Y
INSTALL_PKGS="$INSTALL_PKGS other_package yet_another_one"

# ...

apt-get install $INSTALL_PKGS
```

nyxnor, 2021-04-18:
This is mostly solved. The example above of using a variable when installing packages are a good resource to thin the lines in the build_sdcard.sh and can be implemented.

Kixunil, 2021-04-18:
@nyxnor I can close if you confirm that there are no significant things left from my original suggestion. I didn't follow this project properly for a long time.

Regarding users mentioned in the issue, I create system users with constant name in my project and leave the normal user without modifications. This is more secure anyway.

nyxnor, 2021-04-18:
@Kixunil Maintain this issue open.
* Not all platforms are listed in the build script equally, you can only see [tegra here](https://github.com/rootzoll/raspiblitz/blob/3032410b76b0da248aadb358ab21b7720db3c222/build_sdcard.sh#L146.), but [not present when setting locales](https://github.com/rootzoll/raspiblitz/blob/3032410b76b0da248aadb358ab21b7720db3c222/build_sdcard.sh#L231)
* For now, the after doing the build, it is checked for some essential info, [asking for user confirmation.](https://github.com/rootzoll/raspiblitz/blob/3032410b76b0da248aadb358ab21b7720db3c222/build_sdcard.sh#L176)
There is more to be improved of course.
* And[ this could be shortened.](https://github.com/rootzoll/raspiblitz/blob/3032410b76b0da248aadb358ab21b7720db3c222/build_sdcard.sh#L516). I just think they let this way in a case they want to delete a certain pkg, but still dont make much sense.